martin2250/OpenCNCPilot #76: G-Codes which are not recognized prevent file analysis

deHarro, 2018-11-01:
Hi Martin,

now and then I encounter error messages when loading g-code files generated with various tools. 
They state something like "Found illegal g-code in file at line xx. Processing stopped, run at your own risk."

You already implemented some of those "illegal" g-codes to be omitted, but the list is uncomplete.
Do you see the chance to make those Codes editable, so I can add additional codes to be ignored?

One of the offending codes is "T=xx", tool change, generated by "pcb-gcode.ulp" in Eagle layout editor.

Harald

martin2250, 2018-11-02:
Hi Harald,

do you mean the list for codes which are silently ignored? Good idea, I'll implement that soon (-ish)

Martin

deHarro, 2018-11-02:
Hi Martin,
to be sure, I do not know 8)

I remember that OCP initially had problems with some codes I had in my files (don't know if those from Eagle pcb-gcode or those from Estlcam, both are tools I use) and so you changed something so that it worked afterwards.

I then found this list of codes in the settings window named "GCode" and thought, at least the S command was one of the culprits you had to handle (back then).

So, if you talk about this very list, then yes :-)

Harald

martin2250, 2018-11-04:
Ok, I'll look into that  :)

martin2250, 2018-12-24:
Hi Harald,

I think my latest commit f559415975bdb61c2707df919d928e670fff0413 should cover this, OpenCNCPilot will now ignore some errors while loading files and will instead show a list of warnings. I didn't add a special case for tool change commands (Txx) etc., so they will show up as warning and can (in most cases) be ignored.

Martin

closed 2018-12-24 by martin2250