martin2250/OpenCNCPilot #78: Random Freezing

sirsenor, 2018-11-08:
Hi there Martin!

I downloaded the new version of OCP and tried to isolate some designs. I noticed that OCP randomly freezes and the only way to unfreeze it is to disconnect the CNC physically (pulling the USB cable) and plugging it back in. This freezing is so persistent that not even the task manager can end the process while the machine is still plugged in. Could it be because of the "Work"-Thread blocking the execution(?) of the program.

Best Regards

martin2250, 2018-11-08:
Hi @sirsenor,

that's pretty bad, when exactly does it freeze? while probing or while running the modified gcode? did you change any other settings? could you try the same file+procedure with this and the last working version of OpenCNCPilot? (just run the file in air, no need to actually cut something)
If this persists I'll pull the latest version until it is fixed.

Thanks!
Martin

sirsenor, 2018-11-08:
Im not really sure what triggers the freezing, however it only happened while running some gcode and while the machine was idle (not while probing, at least not yet} . Ill try the previous version of OCP tomorrow and document my results here.

sirsenor, 2018-11-10:
I did some testing with both versions and even the 1.5.2.0 version freezed once, so the problem might be caused by my machine and not OCP. I'll close this for now, cause I got a much bigger issue

closed 2018-11-10 by sirsenor

martin2250, 2018-11-10:
okay, thanks!

 just in case anyone else reads this and experienced a crash: please reopen this issue immediately.