martin2250/OpenCNCPilot #79: Height map not applying correctly

pinomelean, 2018-11-10:
I've been using OpenCNCPilot for some time now to mill the ocasional PCB and it's been without issues for the most part (only human ones up to this point).
This last week i had to make a group of PCBs and when i fired OpenCNCPilot i got the usual update notice, and i upgraded as i always do (to 1.5.3).
When i got to appliying a height map to a loaded gcode and running it, it was milling way too deep into the board. i had programmed it to mill 0.05mm in depth, but it was milling at the very least 5 times deeper.
I tried again two more times thinking something moved during or after probing, and got similar results.
All those times the depth seemed to be consistent (wich is the point of the height map) but i can't measure it to be sure, so emphasis on the "seemed".
Luckily i didn't break any bits, as my feeds are on the conservative side.

After wasting a couple of PCBs triying to get it to work, i downloaded the previous release (1.5.2) and tried again just to make sure.
Surprisingly it worked as well as it allways had, loaded the gcode, zeroed on the surface, probed the area and applied heightmap. it ran well and milled just as deep as i wanted, barely scratching the FR4 composite.
As i had to make several PCBs i kept this version througout, and couldn't test the newer one any further because i ran out of board material.

The 1.5.2 release didn't show the excessive depth problem at any point, but the 1.5.3 did it all three times i tried to mill the ciruit.

It's my first time submitting an issue, so if i've done so in the wrong place, politely tell me so.

martin2250, 2018-11-10:
Hi @pinomelean,

your're at the right place here. I just went over all files that changed between 1.5.2 and 1.5.3 and I could not find any changes that would affect that [(see here)](https://github.com/martin2250/OpenCNCPilot/compare/v1.5.2...v1.5.3). Did you save the heightmap files? If so please share them here and I'll take a look.

Did anything else change from the upgrade/downgrade? specifically the probe feed rate an cause such issues.

Cheers!

deHarro, 2018-11-10:
Hi Martin,
just to dont't let it be missed...
You stated in the changelog:
- > made HeightMap generator use my Calculator library

Perhaps something went wrong when implementing this feature?
Just my 5 cent.
Harald

martin2250, 2018-11-10:
Hi Harald, 
the new code is only used for test patterns in height maps, it doesn't affect normal operation.

Martin

pinomelean, 2018-11-10:
Oddly enough i haven't been able to replicate the issue.
After creating two height maps with 1.5.2 with a homing in between, I've downloaded version 1.5.3 again, and made another two height maps. That makes it four in the same fisical location with a homing operation in between each one.
looking at the files there doesn't seem to be a significant difference, none that can't be attributed to my flimsy machine.
Assuming the 1.5.3 files i used 4 days ago are the same i downloaded today, i cannot explain why it failed three times with three different height maps and all of them to a similar depth. I'm milling a PCB right now with v1.5.3 and it seems to be working just fine.

So unless it happens again or someone else reports a similar problem, consider this issue a "you probably touched what you shouldn't have" on my part.

[hmaps.zip](https://github.com/martin2250/OpenCNCPilot/files/2568763/hmaps.zip)

martin2250, 2018-11-10:
Okay, that's good news. Let me know if you run into any more issues!

Martin

closed 2018-11-10 by martin2250

juggie01, 2018-12-21:
I have just run into this problem, for no apparent reason and after having worked very well for the last few days, instead of cutting, it just skids over the surface. I assumed that something was loose on the CNC (Ozznest C beam) but that isn't the case, there's no discernible play anywhere.

I've also tried re-booting the PC and powering everything down, and then back up - its driving me nuts !!