martin2250/OpenCNCPilot #80: Connection garbled causing machine to turn on spindle

backspace119, 2018-11-14:
Hi, I looked through your other issues to try and make sure I'm not repeating anything here, and I hope I didn't miss one (one was sort of close, but not enough). I have a cheap chinese 3018 Pro cnc with a "woodpecker 3.2" board running GRBL 1.1f. I also downloaded the .Net 4.6.2 installer and found it was already installed

The problem is that when I connect OpenCNCPilot to my machine it turns on the spindle (sounds about like 50% spindle speed) and the machine makes noises like steppers are being stepped but not moving. I previously was using the software that came with the machine, which is candle 1.1.7, and it worked fine for connecting, controlling, and sending GCode (I've made some simpler PCBs with it already).

I was looking to do auto probing for a PCB I'm trying to make, as it has some thin traces, I first found the old GrblHeightProbe2 program (and being dumb and not reading the warning that it was no longer supported I grabbed it and ran it) which actually, interestingly enough, did the same thing that OpenCNCPilot does, but not when it connected, only when I tried to start probing, it also crashed at that time. 

The only other information I have is that the machine reset button doesn't work to get the machine to stop, I have to either connect another software (candle or arduino serial monitor) or I have to power cycle the machine. Also, after connecting to it with the serial monitor it dumps this to console: <8<88888<8888<88888<8888<88888<8888<<8888<88888<8888<88888<8888<<8888<88888<8888<88888<8888<<8
before it shows the Grbl welcome (normally doesn't do this, only after being connected to OpenCNCPilot and having the issue it has)

Let me know if there's other info I can provide.

sirsenor, 2018-11-14:
I have the same machine and was using the board, but my woodpecker had grbl 0.9 installed. Are you sure your board is running grbl 1.1f? If not I would suggest that you update the board first and try to use OCP again.

backspace119, 2018-11-14:
@sirsenor I am certain it is running 1.1f and came with that software initially, I've also reflashed it using the arduino IDE (pretty sure it was successful? in another thread it said to expect errors in the output as it wouldn't be able to verify the program) Here's the serial monitor showing the 1.1f welcome message: 
[image]

backspace119, 2018-11-14:
@deHarro I am certain that the machine came with the 1.1f version, see the comment above. It's always given this welcome message.

sirsenor, 2018-11-14:
> I've also reflashed it using the arduino IDE (pretty sure it was successful?

What did the arduino IDE say? It will either say "Upload successful" or give you an error message.

You could try to flash an EEPROM clearing sketch on the woodpecker board and reinstall grbl again. 

Just copy paste this sketch into a new arduino sketch and upload it. When the clearing of the EEPROM is done the digital pin 13 will be pulled HIGH (on normal arduinos the digital pin 13 is tied to an LED for easy visual inspection of the state of the pin).
`#include <EEPROM.h>

void setup() {
  // initialize the LED pin as an output.
  pinMode(13, OUTPUT);
  for (int i = 0 ; i < EEPROM.length() ; i++) {
    EEPROM.write(i, 0);
  }
  // turn the LED on when we're done
  digitalWrite(13, HIGH);
}

void loop() {
  /** Empty loop. **/
}`

Unfortunately thats the only thing that that comes to my mind.

backspace119, 2018-11-14:
@sirsenor thanks for the quick response, this was the entire output after attempting to reflash grbl 1.1f: 

Sketch uses 30,592 bytes (94%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2,048 bytes.
Low memory available, stability problems may occur.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x47
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x62
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x6c
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x31
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x31

I am a bit worried trying to flash other things on it for 2 reasons, A: it seems to already have 1.1f, so I'm not sure how much good it will do, and B: with it being kind of a cheap board with no docs, I'm worried about bricking it, especially since some of the programming pins are shared with pins for other stuff (or at least I read this in the same thread that was mentioned earlier) Also, I'm not certain that pin 13 has an LED on it, it shows "d13" on the pin header on the side of the board, and I don't see anything else tied to that pin.

backspace119, 2018-11-14:
@sirsenor 

> What you could try is to unsolder the atmega chip on the woodpecker board and solder it to an actual arduino to flash grbl

Ok, I can do this, I was actually just trying to transplant an atmega 2560 from an old mega to one of my 3D printer's boards (accidentally fried it by shorting a thermistor) and the pitch and spacing on those pins are way worse than the 328p, (still waiting on new solder wick to come in today to clean up the chip before turning it on) I'm pretty familiar with a hot air kit although I've not done a ton of SMD work, but the 328p looks easy enough. The only problem is I don't have a nano laying around....I *may* have a cheap chinese uno laying around that used that chip, I'll have to look and see. 

> I meant that arduinos usually have an LED tied to the D13 pin not the woodpecker board.

ah, sorry about that, I misunderstood. 

> This is actually the main reason I threw my woodpecker out of the window (not literally) and designed my own control board for my cnc.

Do you have a link for this? Do you sell them or have the docs to make one? I'd love to get rid of this board, as even compared to some of the boards that came on some chinese 3d printers it looks like garbage.

As a final thought, I have an old Mini RAMBo board from a 3D printer, can I get that up and running for grbl/marlin for a CNC like this?

backspace119, 2018-11-14:
@sirsenor I don't see the link in the last post, hyperlink broke? As far as the mini rambo goes, I'd rather not use it but if I have to I will, I read somewhere that only 328p based boards would work, but if a 2560 will work to that's good news. As far as controlling the spindle seperately, is there a reason for that?

sirsenor, 2018-11-14:
> I don't see the link in the last post, hyperlink broke?

I forgot to paste it and edited in later. Here it is again:
https://easyeda.com/senordoenermann/cncboard

> I read somewhere that only 328p based boards would work, but if a 2560 will work too that's good news.

I actually opened an issue here once asking the same thing and got grbl running on a RAMPS board ( I didnt have my 3018 machine yet back then and was using a homemade DIY 3D printed machine).

> As far as controlling the spindle seperately, is there a reason for that?

I hated the puny little spindle motor that came with the 3018 CNC and bought myself a much stronger and faster one, which - unfortunately - isnt controllable with a simple PWM signal from the control board (Altough I never really looked that deep into it).

sirsenor, 2018-11-14:
> On a side note, did the bigger spindle let you be able to mill aluminum/copper/steel?

I am kind of scared to try it out to be honest and I am pretty sure that milling steel requires a lot more torque than my spindle can provide and much better bits. Engraving may be possible, but I am no expert.

backspace119, 2018-11-14:
@sirsenor alright, I may give it a shot after getting some good upgrades on this thing, but for now I'm just interested in PCBs, thanks for all the help and suggestions, hopefully we can figure out why the woodpecker does this with this software, until then, I'm going to look at trying to reprogram in

backspace119, 2018-11-14:
Ok so an update, I've done what @sirsenor suggested and moved the 328p from a spare arduino (found a chinese uno that used this chip) and it seems to work now (I just programmed the one on the arduino rather than swapping them back and forth. Now, what's weird about this is, the uno now seems to connect to the software just fine as well, without reprogramming. I'm not certain what's going on with this, but I'll leave the issue open so maybe the dev can add this to an FAQ for other people who might have cheap CNCs

sirsenor, 2018-11-15:
Were you able to fix the spindle issue too?

backspace119, 2018-11-15:
@sirsenor sorry for the late report back, everything seems to be working just fine, the board didn't come out well (didn't cut deep enough, all traces are shorted) but it at least was *mostly* level. There were a few traces where it looked like it didn't cut at all, but the majority of the board did just fine. I'm certain with some calibration it will be making PCBs with small traces.

backspace119, 2018-11-15:
So, I left my CNC for a while and came back to it with the intention of moving the z0 down about 0.1mm to try and finish out the cuts, I was jogging the machine with my keyboard to get it to a position where I could do this, and it stopped responding to the keyboard. This happened earlier, so I just disconnected and reconnected, which fixed it last time. This time, it didn't fix it, so I completely unplugged and replugged the machine. Now, it's back to turning on the spindle and having a garbled connection, I'm very confused about this, but there's obviously something weird going on with the connection.

sirsenor, 2018-11-15:
Have you tried a different cable, different USB-port and/or different PC (to make sure those thing arent at fault)?

backspace119, 2018-11-16:
@sirsenor I'll try, but I don't have another laptop laying around and the machine's in an awkward spot to connect to my desktop.

backspace119, 2018-11-16:
I've found the issue, if I connect while the motors/spindle are connected, everything goes nuts, I've confirmed it was also stepping all motors, although some were not moving. Could this be because of shared pins? If so, why does it do it on this software and not Candle?

backspace119, 2018-11-16:
I've done quite a bit of testing, new port seems to have fixed the issue, but we'll see how long this lasts, this definitely seems to be a software issue, although maybe not on OpenCNCPilot's side, we'll see.

martin2250, 2018-11-17:
Hi everyone,

looks like you have your issue resolved, I just want to add a few thoughts:
- one difference between OCP and UGS (maybe also Candle) is that OpenCNCPilot does not reset the Arduino when connecting to the machine. At least UGS enables the DTR line, which resets the Arduino. Preventing this reset allows you to continue a job without loosing your origin, but also allows garbled data to pass through. I don't know why this behaviour started with OCP and then also appeared on other programs, but I doubt that it is any program's fault. This smells like a hardware issue.
- most of your issues should be caused by a bad connection between the computer's USB and the cheap clone, as you stated. Also taking proper care that the noise from the PSU and stepper drivers doesn't interfere with communications is essential.
- Grbl becoming unresponsive when jogging does sometimes happen, I don't know yet why. It is only a mismatch between what OCP thinks grbl received and what was actually received, so OCP thinks that grbl isn't done processing a line, so it waits until it receives the next 'OK'. You don't need to reconnect, just click the progressbar that shows the buffer status to reset the buffer and it should work again. If that alone does not help, a soft reset will fix it every time.

Cheers!

backspace119, 2018-11-18:
@martin2250 Thanks for your reply. On the first point you made, the behavior didn't continue in other softwares, but rather connecting another software would get the machine working again, probably because of the reset that you talked about. On the second point, I believe that it could be it being a cheap clone, but the cheap uno clone seems to work just fine, and from inspecting the circuits the USB to serial seems to be identical (both use the CH340G chip). Then again, on the latter half of your second point, it could very well be noisy lines interfering during initial communication setup. So far, using the new port, it seems to be ok, but the issues last time happened after I'd disconnected and reconnected it later, I'm afraid to do that now because I'd like to get some jobs run through first, but I'll give it a shot soon. To me, this seems like it may be a weird windows 7 driver issue for the CH340G, since putting it on a new port magically fixed the problem. I'll keep collecting more information about this, but it does seem this is probably not an OCP issue, nevertheless, it might be a good idea to put some sort of FAQ in the docs for future people with cheap CNCs like this, just in case someone else goes down this road.

deHarro, 2018-11-19:
Hi guys!
Just want to share some knowledge about my own mill...

I very often read about "cheap clones" of Arduino which seem to have issues in cunjunction with OCP.
My mill works with two Aduino Nano V3 clones coming directly from china.
Both are connected to USB with CH340 chips.
I started with Arduino Nanos consisting of the original FTDI chip but found, that those have to be connected to USB _after_ beeing powered up.
Since my Arduinos are mounted within a comupter case, I have no easy chance to connect USB after powering the Arduinos, so I tried Arduinos with CH340 and found them working well, no issues with power sequence any more.
That said, I have no issues with bad comunication coming from the usage of "cheap clones" or CH340.

The only problem I encountered on the way was when I used a DIY USB connector to contact the USB 3.0 connector on the MoBo. This blue connector has 2 mm pitch and very thin contacts. The pin header I used to make the connection was designed for  standard 2 mm spaced pin headers whose pins are somewaht thicker.
In the earlier times, lets talk about 2 years, all went well, no problems. But thenI got issues like the ones some of OCP users are describing. Stalled connections, aborts in mid of a job, such things.

I wrote, I have two Arduinos in work, the other is for my joystick which at that time had no issues.
So I tested the connection with the other USB port nad the problems were gone.
That found, I ordered a standard USB 3.0 adapter wit hthe original blue connector for the MoBo connector and 2,54 mm pin headers on the other end.
From then on all problems were gone.

In addition I have shielded wires for my steppers and also shielded wires for all connections between mill and computer case, all shields grounded on both ends (no ground wire to earth on the aluminium frame of the mill but the shields electrically coupled to the aluminium frame) to keep electro magnetic disturbances away. In this way the housing of the computer and the mill are electrically coupled with low impedance and the computer case itself is grounded via the mains cord. All power adapters are seated inside of the computer case and are well grounded, each. You may have a look at it on my [homepage](http://www.harald-sattler.de/html/joystick-steuerung.htm) - all in german, sorry, but some pictures. 
Seems to work very well up til now.

Harald

closed 2018-12-22 by martin2250