martin2250/OpenCNCPilot #82: SurfaceScanning, export result as CSV

krtschmr, 2018-12-09:
Is it possible to just run the surface scanning and export the result as a CSV file so we can replicate the surface as a 3D Model in Fusion or somewhere else?

martin2250, 2018-12-09:
Hi, 
.hmap files are actually just xml files. They don't use any existing standard, so you can't just plug them into another program, but a quick python script can convert a height map to any format you need.

krtschmr, 2018-12-09:
ya, i can do sublime and replace it

`<point X="([0-9]+)" Y="([0-9]+)">` with then `$1,$2,` and then replace `</point>` withnothing and it's done.

i wonder why the points are starting from 0,0 instead of negative values, so 0,0 would be the center?

martin2250, 2018-12-09:
I can't remember why I did it that way, this entire file format is pretty horrible. The offsets should be stored somewhere in the .hmap.