martin2250/OpenCNCPilot #89: Interface laggy / low fps when zooming in, only on certain hardware

gamerpaddy, 2019-01-16:
The Interface (viewport) is very laggy when zoomed in.
this only seems to happen on my desktop computer (8700k, 32gb ddr4, 980ti win 10 pro) 
but not on my tablet which i use on my cnc (i5-3320m, intel integrated graphics, 4gb ddr3, win 10 pro) 

video: 
https://streamable.com/khw6i

version: 1.5.4.0

sirsenor, 2019-01-16:
I dont see why OCP would be at fault here, since its working fine on your less performant machine, which is why I would suggest that you update your GPU driver or just be a little more radical and reinstall windows completly.

deHarro, 2019-01-29:
Perhaps this info is of some value for the issue...

I run OCP on an i5 without graphics card, the whole rednering is performed on the integrated graphics of the CPU (not that those integrated graphics kernels are bad in any way. In contrary, I'm very pleased of the overall performance, but I'm no gamer).
I have no performance issues with OPC at all, even when zooming in to a single via fullscreen.

Harald

holance, 2019-01-29:
It depends on the resolution of your monitor as well. If your monitor has HD resolution or connects with multiple monitors, integrated graphics card will not perform well.

deHarro, 2019-01-29:
Ok, I can follow :)
The monitor has only 1280 x 1024 pixels and it's a single monitor installation (at the CNC machine).

On my office machine I use the integrated graphics, too. Here I have two monitors, 2556 x 1440 and 1920 x 1200, respectively. No performance issues with OCP. 
Ok, this machine has an i7@4GHz.

But obviously others have issues with the performance, and trying to improve OCP by changing the graphics subsystem is not so bad, I'm with you.
Harald

gamerpaddy, 2019-01-29:
you guys are mixing this up

-it runs better on integrated graphics on a slow pc
-it runs worse on the one with a dedicated gpu thats overall very fast.  
- but it also runs fine on the gtx 570m laptop which also doesnt have a integrated gpu.
-the one with the dedicated gpu doesnt have a integrated gpu enabled.
-both have the same screen resolution and refresh rate.
-this problem also exists in the examples provided by helixtoolkit-wpf its not a opencncpilot issue.
-removing everything from the 3d viewport except one element still causes this issue, doesnt matter which element is in it.

loading in a gcode file that draws a bunch of lines makes the program unuseable on problematic computer this is all about.

deHarro, 2019-01-29:
I think I should delete my comments above, they seem to add only traffic and leads the discussion into a wrong direction. Sorry.
Please ignore my comments!
Harald

holance, 2019-01-29:
Since Helixtoolkit wpf is using wpf internal 3d engine, you may want to check if you have driver issue.