martin2250/OpenCNCPilot #91: Jogging gets Stuck sometimes, making a full Move

gamerpaddy, 2019-01-28:
Ive had that happen a few times now, but now on the Z axis causing my bit to crash into the part while trying to set work offsets.

Basically when jogging via keyboard sometimes it wont stop when releasing the key, 
it drives the full set-distance then it stops. 
not a huge problem on X and Y with a distance setting of 10mm, but i had this happen on Z causing the not-spinning bit to ram into my part.  

And setting the distance setting to 1mm drives me nuts, clicking 10 times to move 10mm, instead of just holding the arrow key down.

Any ideas?
im using version 1.5.4 and grbl 1.1f

sirsenor, 2019-01-28:
See issue #60 

This is known and the best solution is to use a combination of jogging and controlled jogging (using the arrow keys, pageup/pagedown with CTRL being pressed) and setting a controlled jogging distance of 1mm or something similar.

martin2250, 2019-01-28:
Hi, I've experienced this before, but I can't pinpoint the cause. That's why there are the jog distance settings in the first place. You should always set them low enough where the machine does not crash even if the full move is executed. I know that's quite annoying, but right now this is the only way to jog safely.

gamerpaddy, 2019-01-28:
cant you just send the jog abort command multiple times in case the first one didnt arrived in one piece?

martin2250, 2019-01-28:
Sure, but that wouldn't fix the problem. The problem is that the key presses (or more specifically the release) is not handled correctly by the UI. I'll have to do some further testing to see if this is a problem with OCP's code or the underlying UI framework.