martin2250/OpenCNCPilot #93: Can't see serial port on 1.5.4 (but it does on 1.5.3)

Sheenygami, 2019-03-10:
Hello Martin,
first of all, congrats for this great work!
Unfortunately I have the following issue with version 1.5.4 and my Arduino Nano CH340 based cheap chinese 3018 router. When I plug the cable in, the COM port doesn't show up in the listed serial ports. I just get an empty list. Running as administrator doesn't fix.
At first I naively tried re-installing CH340 drivers and upgrading the Grbl firmware to the latest version but none of these could solve the problem. Then I tried downloading OpenCNCPilot 1.5.3 and in there the COM is correctly detected.
Relieved, I will just keep using this version, but I thought about reporting the issue, hoping that it's a just a minor bug that could be fixed in the next release. 

Thanks for your attention and have a good day!
Andrea

martin2250, 2019-03-10:
Hi Andrea,

thanks for reporting this. In 1.5.4 I changed the serial port selector to use the Windows Management Interface instead of the old SerialPort.GetPortNames(), so I could add port descriptions (896df76e42474134e37cc510c270c252873483f6).
Two questions:
- What version of Windows are you using?
- Does this happen only with CH340 based boards? Could you try to see if an official Arduino or FTDI adapter are listed in 1.5.4?

Until I've fixed it, you could try this workaround: once you've selected the serial port in 1.5.3, you should be able to upgrade to 1.5.4 and the port should remain active.

Best,
Martin

Sheenygami, 2019-03-10:
Hi Martin, thank for the feedback!
I'm using Windows 10 Home 64-bit.
I was facing this problem with both Arduino Nano and Arduino Uno (both CH340 based), unfortunately I don't have any official board... 
But it seems that your workaround has worked! After using my nano with version 1.5.3 today in the garage, when I came back to my studio and read your reply, I tried to open 1.5.4 and I could see both Nano and Uno. So it seems like I can't reproduce the problem anymore, as now it works fine with both versions! 
Is this some Windows black magic? 
Thanks a lot anyway for the support :)

martin2250, 2019-03-11:
Hi Andrea,

that's very strange. I didn't expect it to enumerate them both after upgrading, just remember the one which was selected. Oh well, that's Windows for you :D
I'll close this issue, if the problem reappears, just reopen it again.

Cheers!

deHarro, 2019-04-01:
Hi Martin,
I just tried to connect my office PC to my test equipment (a bare Arduino Nano no motors attached, GRBL 1.1f installed and responsive (checked with other G-code streamers)) and found the Arduino not recognized.

Win 8.1, 64 bit - OCP V1.5.5.0 - Arduino with CH340

I'm not sure whether I had an older version of OCP already running with this Arduino, will check that (for your work around). I think I changed the USB port recently...

Harald

Oh yes, please reopen this issue - don't know how to do it myself.

martin2250, 2019-04-02:
Hi Harald, 

thanks for pointing that out, I'll try to fix it ASAP!

martin2250, 2019-04-02:
Should be fixed now. If you can confirm it with your board, just close the issue again :)

deHarro, 2019-04-02:
Hi Martin,
that's ASAP indeed ;-) Thanks for the really fast fix!
...
Just tested it (on the office PC):
We're half way through.
The port is now listed correctly, but the communication between GRBL and OCP is dysfunctional.
Again OCP claims "Disconnected" instead of "Idle" and "GRBL Settings" reveals in an empty Dialog.

Now I see this wonderful new checkbox in Settings: "Reset GRBL when connecting".
When checked, the communication is established and all works as expected.

BUT...
I tried the startup process several times, with and without this checkbox checked.
Most (now every?) time it works flawlessly.
I have the feeling that only the very first startup of the newly installed OCP version has the issue with remaining in "Disconnected" state, since the last 5 to 9 starts are made without the checkmark in "Reset ...".

Reverting to version 1.5.5 and back to 1.5.6 didn't change anything, the communication works stable without the checkmark.

I'll keep an eye on this topic and will surely inform you if I encounter problems when moving this version to my mill :-)

Thanks!
Harald

deHarro, 2019-04-02:
Mhmm, please enlighten me!
How do I close (or reopen) an issue? I have no clue :(
Thanks!
Harald

deHarro, 2019-04-02:
close

deHarro, 2019-04-02:
close #93

martin2250, 2019-04-02:
If you have the permission to open/close issues (not sure about that) , there should be a 'close/reopen' button next to the comment button.

Martin

closed 2019-04-02 by martin2250

deHarro, 2019-04-02:
OK, there is no button other than Comment, so I'm fine.
I feared I am too dumb to see the obvious way... ;-)
Thanks!
Harald