martin2250/OpenCNCPilot #95: orientation of the grid to be defined as default

deHarro, 2019-03-14:
Hi Martin,

I once again have a somewhat unusual wish for improvement of your fine software.

As already mentioned from time to time, my mill is oriented unusually (forgive me, this picture is taken in a very early state when I used to control my mill with SerialComCNC, nowadays I always use OpenCNCPilot :-)

[image]

Zero is front right, X-axis is pointing away from me, Y-axis pointing to the left.

So I always have to adjust the grid orientation in OCP according to get the picture of my workpiece optically aligned to the reality (preventing me from getting too much headache in trying to do this alignment on my mind - did I already mention that I had much fun in the past when I mixed up this correlation? ;-)

So, getting to the point...
Can you imagine to think about spending an option to define the initial orientation of the grid for me?
Would be very fine... :-)

Thanks, Harald

martin2250, 2019-03-15:
Hi Harald,

this should be simple enough, I'll add it to the next release :)

Best, Martin

deHarro, 2019-03-15:
Hi Martin, 
fine! :-)
I already looked up a possible location... "settings->viewport" ;-)
(having in mind that you are unhappy in designing new dialogs)

Harald

commit 53862026 (2019-03-21): add 'save viewport' button

close #95
closed 2019-03-21 by martin2250