martin2250/OpenCNCPilot #96: Spindle immediately starts when connecting

pilant, 2019-03-16:
I'm trying to use OpenCNCPilot v1.5.4.0, and am having a number of problems.  At the moment,
when I connect to the COM port, the spindle immediately starts up and the table starts to move
slowly (Y axis).

The puzzling thing is it did this for the first couple of times I started the application, then for unknown reasons stopped.  The spindle would not start up when I connected (and opened the ngc file).  After "connecting" correctly several times, it is now back to starting up the spindle.
I did try to open the ngc file prior to connecting, and the spindle still started upon connecting.

I tried running Candle 1.1.7 (with grbl 1.1f) and everything works fine when connecting.  So I don't expect it is the grbl board.

One possible clue is I have had to start / stop / abort OpenCNCPilot several times, so ti may be possible there "something" left around causing this (registry, config file, etc.)

Any ideas?  (Am I doing something wrong?)

pilant, 2019-03-18:
BTW, I tried enabling "Show Status Messages" and "Log traffic to file".  There were no status messages generated, and the log file was empty.

pilant, 2019-03-18:
I installed a serial port sniffer (Free Serial Analyzer from HDD Software) and am seeing some really strange things.  When I start up Candle (to reset the home position and raise the tool after probing) I get the following:
 (WRITE to grbl)
000138: 2019-03-18 17:56:26.6639570 +0.0048308

 18                                                .
000156: 2019-03-18 17:56:28.1296627 +0.0339690

 3F                                                ?

(READ from grbl)
000142: 2019-03-18 17:56:28.0926522 +0.0000088

 0D 0A 47 72 62 6C 20 31 2E 31 66 20 5B 27 24 27   ..Grbl 1.1f ['$'
 20 66 6F 72 20 68 65 6C 70 5D 0D 0A 5B 4D 53 47    for help]..[MSG
 3A 27 24 48 27 7C 27 24 58 27 20 74 6F 20 75 6E   :'$H'|'$X' to un
 6C 6F 63 6B 5D 0D 0A                              lock]..
000160: 2019-03-18 17:56:28.1328343 +0.0000046

 3C 41 6C 61 72 6D 7C 4D 50 6F 73 3A 30 2E 30 30   <Alarm|MPos:0.00
 30 2C 30 2E 30 30 30 2C 30 2E 30 30 30 7C 46 53   0,0.000,0.000|FS
 3A 30 2C 30 7C 57 43 4F 3A 31 34 35 2E 36 30 36   :0,0|WCO:145.606
 2C 32 2E 38 37 33 2C 2D 32 2E 34 35 31 3E 0D 0A   ,2.873,-2.451>..

When I start up OpenCNCPilot and connect I get the following:
(WRITE to grbl)
045002: 2019-03-18 17:48:51.5075184 +0.0003475

 0A 24 47 0A 0A 24 23 0A                           .$G..$#.
047935: 2019-03-18 17:48:52.1083452 +0.0011208

 3F                                                ?

(READ from grbl)
045007: 2019-03-18 17:48:51.5085616 +0.0000038

 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38   8888888888888888
 38 38 38 3C 38 38 3C 38 38 3C 38 38 3C 38 38 38   888<88<88<88<888
 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38   8888888888888888
 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38   8888888888888888
 38 38 3C 38 38 3C 38 38 3C 38 38 3C 38 38 38 38   88<88<88<88<8888
 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38   8888888888888888
 38 38 38 38 38 38 38 38 38 38 38 38 38 38 3C 38   88888888888888<8
 38 3C 38 38 3C 38 38 3C 38 38 38 38 38 38 38 38   8<88<88<88888888
 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38 38   8888888888888888
 and the response to the second write is more of the same.

So it would seem in the case of OpenCNCPilot, the grbl software is somewhat "confused".  Baud rate for the port is correct in both cases, but I don't know about the size, parity, and stop bit settings.  Also if there are any RS-232 control signals (DTR, CTS, etc.) coming into play.

Maybe this will help.

- Mark

pilant, 2019-03-18:
I forgot to add, after getting the strange behavior running OpenCNCPilot, if I start up Candle, everything works after I clear the alarm.  I do not know the source of this alarm.  (You can see alarm in the Candle "sniff".)

martin2250, 2019-03-19:
Hi Mark,

thanks for looking into this with the analyzer. This is some really strange behavior. The only difference in that regard could be that OpenCNCPilot does not set the DTR signal, so grbl is not reset after connecting (don't know if Candle does it too). What board are you using, what USB-serial converter is used? Do you have a spare Arduino lying about by any chance? I'd be interested to see if it also happens with that.

Martin

deHarro, 2019-03-19:
Hi Martin, Mark,

I stumbeled over this magical "3F" you encountered in the trace, when dealing with another G-Code streamer (GRBL-Plotter), see [here](https://github.com/svenhb/GRBL-Plotter/issues/34#issuecomment-419526212).

As far as I remember, character 3F (65 dec, '?') replaces a character beyond 127 dec, when using the standard ASCII encoding in serial communication.
The source of those chars beyond 127 dec was the "Jog Cancel" command, which is 0x85.
This command lies beyond the magical border of 127 (decimal) up til where "normal" characters are treated well by the usual serial communication routines.

Perhaps you, Martin, chrashed the communication by switching to another means of serial communcation  scheme (I think I read this in some of the other issues? But perhaps I get this wrong...).

Sven (of GRBL-Plotter) had to replace the receiving code from 
`rxString = serialPort.ReadTo("\r\n"); `
to be
`byte[] rxBuff = new byte[serialPort.BytesToRead];
serialPort.Read(rxBuff, 0, serialPort.BytesToRead); `

See [here](https://github.com/svenhb/GRBL-Plotter/issues/38#issuecomment-437573069) for the full byte handling code he implemented.
At the top of the linked thread you see the former code which throws the '?' (3F).

HTH,
 Harald

pilant, 2019-03-19:
Martin, I just checked the Candle source code, and the serial port is configures as 8-N-1, with no flow control.  (So RTS/CTS will not be used.)  I have some other Arduino (Uno)  boards, but I'll have to check to see if the code will work correctly on just a bare board; one without the stepper drivers.

Looking at the OpenCNCPilot sources, it appears the setup of the SerialPort class is pretty standard.  Although I didn't see anything in the class reference documentation to indicate the default for the RtsEnable property.

Harald, the "?" is used by Candle (and grbl 1.1f) as a board query command.  (See the Wiki: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands)  In the Candle sniff above, this is the "alarm" information in response to the "?".  Looking at the link, it seems the GRBL-Plotter log windows are where the 0x85 is getting translated to 0x3F.  (The sniffer traces show the real 0x85 in the stream.)

- Mark

pilant, 2019-03-19:
Martin, the board is a Woodpecker CNC board, which uses the CH340 USB to serial chip.

BTW, I just tried the Arduino IDE as well as PuTTY to connect to the board using the CON port, and neither started up the spindle.  I was able to send G-code commands to the board without any problems.  (And they actually worked :-)

- Mark

deHarro, 2019-03-20:
Hi Mark,
ok, thanks for the explanation.
I have to confess, I didn't dive too much into the trace, only was triggered by the value itself.

Harald

pilant, 2019-03-20:
Some more information.  Looking at the Woodpecker schematic, the CH340 offers TxD and RxD (as expected) but also DTR.  All of these are connected to the ATMEGA328P chip.  Worth noting, however, is DTR being used to toggle the RST pin.

Setting a breakpoint at the port.Open() call, I started OpenCNCPilot.  As expected, when I attempted to connect the com port, I hit the breakpoint.  Looking at the fields and properties I didn't see anything unexpected.  When I continued from the breakpoint, the spindle started up.

On a whim, I restarted the program and decided to set DtrEnable to true after hitting the breakpoint.  (Before calling port.Open().)  In this case, the spindle did not start and I was able to use the program as normal.

Looking at Qt, which Candle uses for portability, QSerialPort appears to set RTS and DTR by default.  I'm not sure what the Arduino IDE and PuTTY do.

So, it appears there is some interaction with the OS (Windows 7 Ultimate in my case) the com port, and the Woodpecker board.  Since my RS-232 breakout box won't work with the CH340, I may have to hook up an oscilloscope (I don't have a logic analyzer :-) to the DTR / RST pin on the Woodpecker to see what is really going on.

What is also unknown is why the program wasn't "working" (spindle started turning), then started "working" (spindle didn't start turning), and then went back to "not working".

- Mark, still scratching my head

martin2250, 2019-03-21:
Yes, DTR is commonly connected to reset. The Arduino IDE also sets this, PuTTY can be configured to set it as well (don't know about the default).

My guess would be that grbl receives some garbage while it's not connected and when OpenCNCPilot starts, it sends a few newline characters which make it execute whatever it received. All other senders flush the rx buffer by resetting the board. Is this maybe some quirk with the CH340? (not sure why it would always be spindle enable and not some random action though).

If you're good with soldering, you could try to add a pullup resistor to the AtMega's (grbl's) RX pin, that way it won't be as sensitive to EMI should the TX output of the CH340 be tristated.

I think I'll add an option to enable DTR (off by default of course), this seems to me like the easiest fix.

deHarro, 2019-03-21:
Hi Mark!
> ...but also DTR. All of these are connected to the ATMEGA328P chip. Worth noting, however, is DTR being used to toggle the RST pin.

Are you sure that DTR from the CH340 is connected _directly_ to one of the AtMega328 pins?
The "normal style" has a capacitor in series between CH340 and the 328 (C6, bottom left on the sheet).
This capacitor issues a spike on RST at the 328 but prevents a steady signal.

I found a [schematic of the woodpecker board](https://github.com/gnea/grbl/issues/123#issuecomment-358299534). There this serial capacitor is implemented as expected.

Harald

pilant, 2019-03-21:
Martin, it is really strange that I appear to be the only one seeing this; unless there is something else (registry setting?) at work.  In any event, I was thinking a "DTR" check box in the settings would be a fine way to go.  (I'll just use my modified code until you determine how you want to address this issue.)

Harald, the DTR line from the CH340 does indeed go through a DC blocking capacitor (C6) to the 328.  I was not sure how technical to get.  (I've been doing software and analog/digital hardware for 40+ years, but I realize it can be *really* foreign to a lot of folks :-)

I know of two versions of the Woodpecker schematic.  One dated October 2016 and another dated October 2018.  Can't remember exactly where I found them as I copied them locally.

With this sorted out, all I have to do is figure out why the tool isn't cutting through the PCB copper even after applying the height map.  It is always something :-P

- Mark

deHarro, 2019-03-21:
Hi Mark,
ok, ok, I see... :-) And this sort of stuff is not so foreign for me, I'm in electronics for 40++ years now :)
I only pointed out topics that seemd to be relevant to me in the given problem area.

Perhaps I should give a short explanation why I'm interested in your problem:
I have two Aduinos working in my CNC controller. One gives GRBL 1.1 its home, the other generates step impulses according to the throw of two thumb joysticks to position the router head in X,Y and Z (sadly no logical connection between the joystick and OCP, they don't know each other).

When I started with my CNC router I decided to build the controller part into a standard PC housing.
So I have no further access to the USB ports of both Arduinos what was uncool because I found that the communication between my controller software used back then SerialComCNC), only works when I first start the PC and afterwards give power to the Arduinos. No chance in the closed PC housing. Those Arduinos had FT232 chips on board.
Happily it turned out that Arduinos with CH340 had no such restrictions on power sequencing, so I changed the Arduinos and had no problems from then on.

My CH340 equipped GRBL Arduino is working with different control software with no issues:
OpenCNCPilot, SerialComCNC, GRBL-Plotter, Estlcam

Ah yes, to be complete:
I have working configurations of Arduinos with GRBL 1.1f and CH340 with different versions of OpenCNCPilot under Win 8.1 and Win 10.
So in my eyes there is no obvious incompatibility between OCP and CH340 equipped boards (despite the fact that every now and then this topic arises in the issues of OCP).

Hmm, not so short as expected... ;)

Anyway, keep us informed when Martin launched the DTR option...
Harald

commit b2d42294 (2019-03-21): add DTR enable option

fix #96 (more like a workaround, but if it works...)
closed 2019-03-21 by martin2250