martin2250/OpenCNCPilot #97: Auto levelling doesn't appear to be working (for me)

pilant, 2019-03-16:
During one of my successful connection attempts (issue #96) I was able to open the ngc file, create a new height map, and apply the generated height map.  However, when I starting to actually cut the traces, only about the left quarter of the PCB had any traces cut.  For most of the rest of the PCB, the tool never even touched the PCB.

The only thing I noticed was the Z axis (machine?) position indicated in the plot area seemed "off",  With a little experimenting, I noticed each time the "Apply HeightMap" button was pressed, the Z axis position changed.  I couldn't find any explanation for this, so don't know if it is a bug, feature, or something else :-)

I believe I probably hit thie "Apply HeightMap" button more than once before actually starting to cut the traces.

Any ideas what I might be doing wrong?

martin2250, 2019-03-18:
Hi,
the toolpath is shifted by the height map every time you press 'apply height map'. This means that pressing the button twice will overcompensate.
There is no safeguard because that way you can use the 'generate height map from equation' feature to manually adjust the toolpath.

I'll look into your other issue soon

closed 2019-03-18 by martin2250

pilant, 2019-03-18:
Is there any way to undo this over compensation?  Or is it necessary to redo the height map generation?

deHarro, 2019-03-18:
Hi!
1. You can save the height map to a file and reload it and apply it again.
2. If you have just probed the map and applied it twice, you can reload your design from file and apply the map again (provided you kept OCP running and you didn't discard the map).

HTH
Harald

pilant, 2019-03-18:
Thank you.

- Mark

martin2250, 2019-03-18:
@deHarro

regarding point 1: just reloading the height map won't reset the toolpath. you have to re-open the gcode file.

deHarro, 2019-03-18:
Jepp, thanks!
I'll preferably should edit my comment to get it right
...
done.
Harald