martin2250/OpenCNCPilot #102: upgrade grbl

miranmn, 2019-04-01:
Hi there and thanks for using OpenCNCPilot,

before you submit this issue please make sure to check the following things:
- you're using the latest version of grbl (1.1f)
- you have the most recent version of the [.NET Framework](https://www.microsoft.com/net/download/dotnet-framework-runtime) (minimum 4.7)

If you still need help, delete this text and describe your problem and I or another member of our small community will help you.

miranmn, 2019-04-01:
Hi there,
I need a step y step tutorial to upgrade my usb board port v1.0.9.15 to v1.1f to be compatible with autoleveller CNC PILOT. Or someone can tell me if the board with version 1.1j will work with cnc pilot?

martin2250, 2019-04-02:
For instructions on upgrading grbl, see the grbl wiki: https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino#for-windows

closed 2019-04-02 by martin2250