martin2250/OpenCNCPilot #103: CNC pilot version 1.55

miranmn, 2019-04-02:
Hi, Can you tell me if this version 1.55 of CNC pilot, will work with USB control board GRBL
version 1.1 j ?

martin2250, 2019-04-02:
There is no grbl version 1.1j. The latest release is 1.1f. I don't test compatibility with other versions of grbl, so you'll have to try it out.

miranmn, 2019-04-02:
Hi, thanks for the feedback, but look the following link to see the board grbl 1.1j
https://www.ebay.com/itm/GRBL-1-1-J-USB-port-cnc-engraving-machine-control-board-3-axis-control-laser/254043095018?hash=item3b2625ffea:g:ZHQAAOSwXI1cJkxP&frcectupt=true
I am thinking to buy it to replace my old 0.9.15 board!

martin2250, 2019-04-02:
Idk about the board. If it still works just update the firmware on your old one. grbl 1.1j does not exist (yet), maybe the ebay seller modified it?

closed 2019-04-12 by martin2250