martin2250/OpenCNCPilot #104: height probing issues

deHarro, 2019-04-14:
Hi Martin,
I decided to open an issue on its own for my research in probing with OCP.
Again, please don't be offended by my writings! I will happily let me convince that I did something wrong and OCP is right.

So, what did I do and especially, what for?

In short: 
I think a CNC router should carve trenches exactly as deep as it is defined in the g-code.

The long version:
I realized some weeks or month ago that my height maps didn't result in clean layouts right away after probing and applying the map to the g-code. As often, I thought, ok, I did something wrong, corrected the height adjustment manually and got a decent layout to work with.

Then, two days ago, you mentioned, that you have issues with that, too. That makes my bells ring and I started to investigate on that.
Background, as stated above: Our router mechanics and your code should be able to maintain the reckoned deepness of the trench under all conditions that might arise.
Will say, **if I set the carving depth to 0.04 mm and OCP probes an offset of 0.02, reckons this offset into the g-code, then the resulting trench schould be 0.04 mm away from the surface of the copper clad.**

In real world my router scarcely scratches the surface of the copper clad when operated as stated above. So I think, there is some possible improvement and I want to find the reason.

You emphasized that you are sure, the mathematics in OCP are well done, what I believe is true as far as you checked it :-)
You probably have checked your proposed way of testing (" try generating a height map with a formula (see the bottom of the "New HigheMap" dialog) and make the unevenness large enough to see by eye") over and over again to be sure on that.

So, having doubts about the correctness of the process in mind, I went another way and included the process of probing into the test chain.

I soldered two pieces of board onto another board to get two distinct heights for probing:

[image]
[image]

Then I mounted my pogo-style testpin (spring loaded to prevent accidentially diving the pin into the board) and started probing a very small area, just big enough to have space for four holes with Z=0 as target depth:

[image]

The settings for probing are this:

[image]

(I know, the plunge speed is too high, but for this test with my sping loaded bit it's ok).

The resulting heightmap is this:

[image]

When I apply the map to my g-code I get this:

[image]

It's not clearly to be seen in the picture above, but perhaps you can spot the two lines which mark the movement of the bit going way into the doubled board. This came from line 5 and line 7 in the code, respectively (red circle).

This is an error(?) of Estlcam, I suppose, which I was not hunting for but stumbeled over accidentially. I have to investigate on that, don't know if this is a parameter I can adjust in Estlcam.
Estlcam goes to 0.5 mm above the board with "G00 Z0.5000", which obviously is not handled by OCP, before doing the actual hole with "G01 Z0.0000 F100 S12000", which OCP adjusts correctly.

So far for Estlcam, now back to OCP...
If I remove those offending G0 Z0.5 commands from the g-code, my test bit just touches the surface of the boards on its different heights, all is well and safe. The overlayed (g-code and map) heights are absolutely sane and as expected from the values of the map and the g-code.

My theory of a possible error in OCP is busted so far.
Unclear for me is the deviations when doing a layout, still.

Harald

deHarro, 2019-04-14:
I did some additional tests with holes diving into the material (such as when carving a layout).
Here no obvious errors either.

I surrender and apply some µ manual as you proposed.

Harald

deHarro, 2019-04-15:
Back again... more tests done.

I didn't pay attention to this fact, but retrospective it was always there.
The only deviation which I could always spot is a difference between Z=0 I got by issuing the following macro I defined
```
G38.2Z-10F20
G92Z0
$J=G91F100Z1
```
and the probing value, derived at the identical home position (X=Y=0).
When probing at home position I always get a reading of ~0.02 (exactly 0.019 .. 0.022).

If I take this into account, I reach the magnitude of deviation we both get when carving our boards (remember, I started arguing with this 20µ offset I get when carving).

Lets reckon: I have to adjust my Z axis two to three clicks down from probed Zero.
2..3 clicks, 200 steps/revolution, lead 3 mm/ revolution results in ... 40µ .. 45µ.

Ok, that's heavy, compared to your proposal of 5µ increments, but when we take into account that you started with 0.05 as depth and my depth value was formerly 0.04, the difference is getting smaller.

But back to the facts:
Can you please explain, why my macro (see above) and your probing algorithm come out with different values for Z=0?

Harald

sirsenor, 2019-04-15:
> When probing at home position I always get a reading of ~0.02 (exactly 0.019 .. 0.022).

I also tested this once and came to the same conclusion. Somehow the probing is always off by some microns and it definitly has to with software (not sure if GRBL or OCP).

> I'm pretty sure that if I substract these 20µ from every Z value in my g-code file, I will get the desired result right from start after applying the probed map.

Thats exactly what I did for the last few months. After routing at -50μm and not getting enough depth I just decided to route at -80μm and the boards came out perfect. It doesnt bother me tough, because once you measure and test the desired routing depth and generate your gcode accordingly, the boards will look very good and it does not matter if the probing is inaccurate. It's reliably inaccurate and the error can be adjusted.

deHarro, 2019-04-15:
Hi Can,
ok, thanks for the confirmation!

I myself am sort of perfectionistic but sorrily not able to make such fine software on my own. But I'm confident that Martin will hunt down the problem, just to get this deviation out of his program :)
Or he at least will explain, why this has to be solved in exactly this specific way...

Harald

martin2250, 2019-04-15:
Hi everyone,

this difference between manual and automated probing comes from the way grbl does probing. grbl doesn't stop immediately when it senses contact, but rather decelerates in finite time. This means that the position where the contact was closed (which is stored in the height map) is not equal to the position where the machine comes to a stop (which your macro uses as Z=0).
This all assumes, of course, that you always use the same feed rate for probing. 

After applying the height map, the height of the toolpath is the desired depth relative to the contact point. The work coordinate system doesn't matter here at all, you could set your Z origin at one millimeter above the copper and it would still work fine.

If you consistenly have to mill deeper than your original 0.04mm, just increase it. At least in this regard, OpenCNCPilot is not at fault :)

Martin

deHarro, 2019-04-15:
Hi Martin,
thanks for the explanation. As I read it, it is totally clear to me and I will circumvent the problem as Can (@sirsenor) described it.

Harald

closed 2019-04-15 by deHarro

martin2250, 2019-04-15:
Just another tip: OpenCNCPilot saves the probe position to macro variables, so you can use something like this for your probing routine to set Z=0 at the actual contact point:
```
G38.2 Z-10 F20
G92 Z{WZ - PWZ}

```
I'm on my phone right now, so this might not work exactly as shown, but you get the idea. Of course this only works if you enable expression expansion (not sure what I actually called this) for this macro.

Martin

deHarro, 2019-04-15:
I'll give it a try, thanks.
Your macro variables are explained anywhere?

Harald

martin2250, 2019-04-15:
see https://github.com/martin2250/OpenCNCPilot#manual-expressions

deHarro, 2019-04-15:
Ah, I recognize that I read that once upon a time... didn't know what to do with it and forgot about it :-)
Thanks again!