martin2250/OpenCNCPilot #105: Autolevelling doesn't act properly

mdsultan11, 2019-04-26:
[image]

Hi Martin,

I'm facing issue while uploading the code to OpenCNCpilot. I have gernerated my PCB gcode using Flatcam. yo can see the error issue in picture.
And one more issue is that my CNC is not making a track in the place of red marked area in height map.
please solve my problem as soon as possible.

Thanks and Regards 
Md. Sultan

martin2250, 2019-04-28:
Hi,

from what I can tell you can ignore these warnings, no worries there. I should probably get rid of the 'zero-length' warning.

As for your other problem: this sounds a lot like the PCB is not glued down properly, so it can bend under the pressure of milling (but not during probing, which is why it still shows up as red). If you aren't doing so already: use a stable wasteboard (I'd recommend plywood at least 9mm thick) and use double sided tape to glue the PCB down over the entire surface.

Best
Martin

mdsultan11, 2019-04-29:
Hello Martin,

Thanks for the Clue you have shared with me.
I will come back once i check with the PCB cutting using double sided tape.

Thanks and Regards 
Md. Sultan

mdsultan11, 2019-05-02:
[image]
[image]

Hello Martin,

As you said i ignored the error i will get in the beginning while dumping the code to OpenCNCpilot and also i glued it my PCB plate using double sided tape to the plywood waste-board, but then also you can see the Red area occurred on height map done by probing and also you can see the actual PCB cut has not occurred in the same place(Red area) in another pic.
Can you please tell me the reason for this and solve my problem as soon as possible.

Thanks and Regards
Md. Sultan

deHarro, 2019-05-02:
Hello mdsultan,
from the picture it seems that only one single point is probed wrong.
Questions to your findings:
- does this error occur every time you do a probing session?
- is it always at the same position related to your waste-board, or does the error positon move around from run to run?
- is there the chance that your probing contact gets shorted to ground of your router at any other point (cable sqeezed or similar fault)?

To get a quick remedy for you, you may safe the height map after probing, open the map file in an editor and find the one probe point which is significantly out of sequence and edit this point to the middle value between the two adjacent points. This should "heal" this deviation for the given probing run and you will be able to mill your board without the error.

Harald

luizabbadia, 2019-05-02:
Hi, all.

I had the same problem some time ago, and I can remember because your
problem has the same symptom as mine.
If you observe, the round tracks that represent the area where the IC is
fit, they are not perfectly round.
I found this problem after rechecking all the motors connections, ( they
were not good, poor contatcs, bad connections etc...).
Those "not-round" elements were the consequence of poorly connected XY
motors and, off course, depth variances were caused by bad connections to Z
motor.
Weel I hope this is helpful

deHarro, 2019-05-02:
Hi Luiz,
I think, that is not the problem of mdsultan.

But to emphasize on the not round tracks, another reason for such results is mechanical backlash at one or both axis (X and Y).
Since all of the (not) round traces in mdsultans picture are shaped very equally, I think this is more likely the reason for this sort of problem.

But again, backlash on the Z axis (as I understood what you wrote) wouldn't result in one sharp jump at one point. 
The reason for his problem must lie elsewhere IMHO

Harald

closed 2019-05-02 by martin2250

deHarro, 2019-05-02:
Hi mdsultan,
glad I could help :-)
 Harald

mdsultan11, 2019-05-07:
> Hi mdsultan,
> glad I could help :-)
> Harald

Hello Harald,

I have small Openbuild CNC machine which 'm using for PCB cutting, as you have mentioned the round tracks in PCB are not coming properly because of mechanical backlash, can you please give a solution how can i get a reduce the backlash in my CNC machine. 

Thanks and Regards
Md. Sultan

deHarro, 2019-05-07:
Hi mdsultan,
my machine has threaded rods and plastic nuts.
What is your type of transmissioin? lead screw and nuts or pulleys and belts?

If you have lead screw and nuts you probably have the option of two nuts which are to be countered / locked against each other or against the base from one side and the other to minimize the backlash.

The force of this counter against each other has to be high enough to minimize the backlash but not too strong. The screw must turn easily still.
Normally there is a means of locking the two nuts with headless screws or something similar.

If your machine is driven with belts, then the backlash depends solely from the tension of the belts, provided the stepper motors drive the blet pulley directly (without gearing).
If there are gears, their mechanical play adds to the backlash.

HTH
Harald