martin2250/OpenCNCPilot #106: Laser Engraving with OpenCNCPilot

rynaldi95, 2019-04-30:
Hi,
I am Rynaldi. Currently, I am working on a laser engraving machine project. The part that will be engraved is a skate-board which has an uneven surface. Therefore, I need to be able to control the Z-axis in order to have the same focus of the laser while engraving on a curve. Previously, I used the LaserGRBL software. However, currently there are no Z-control axis and the developer referred me to use OpenCNCPilot as it has this Z-control. Can this software be used for laser engraving actually?

martin2250, 2019-05-01:
Hi Rynaldi,

that would depend on the GCode generated by Lasergrbl. If it only uses M-codes and/or S (spindle speed) commands to control the laser, I see no reason why it wouldn't work. If you're unsure, you can post an example gcode file on pastebin and share the link here.
You'd need some kind of switch to probe the non-conductive wood surface, @deHarro has a tutorial for modifying a 3D printer levelling probe for this kind of work. Alternatively you could also cover the wood in a conductive material like aluminum tape.

Martin

deHarro, 2019-05-02:
Hi rynaldi,
if interested in the sensor you find the modification described [here](http://www.harald-sattler.de/html/3dtouch_sensor.htm).
The site is written in german, but I provided a (or at least some sort of) translation feature (last topic in the navigataion bar on the left).
A word of caution: The described modification is not easy to perform and you may destroy the sensor accidentally.

Harald

closed 2019-05-10 by martin2250