martin2250/OpenCNCPilot #107: probing along a (to be) given line...

deHarro, 2019-05-02:
Hi Martin,

I again have a very special problem to solve, naturally concerning probing and carving with OCP ;-)

Short: Does your algorithm support one-dimensional height "maps"?
Long:
You already realized my activities in model train things.
Now I face a well known problem when dealing with ROKAL locos and "modern" rail material.
The original ROKAL rails where made of some sort of cardboard as sleeper (Schwellen) material, the rails itself (die Laufschienen) are formed of thin metal sheet material. These old rails are very crude when it comes to running characteristics and back in 2004 I decided to exchange all of those old rails with new ones from Tillig or BTTB.
These modern rails for TT-sized models are made up of plastic sleepers and again metal sheet as rails but the appearance and performance is far beyond.

So far there is no problem at all, but now we come to crossings and turnouts...
Here we face the problem that at some point on rolling over the crossing, the wheels have to be guided over areas where there is no leading side for one of the wheels.
To circumvent this problem, the clever engineers developed guidances which prevent the axle from deviation by guiding the axle on the inner radius of the opposite wheel.
This postulates a given distance between the inner faces of the two wheels of one axle.
And here we encounter the problem: ROKAL locos (and wagons) have 8.5 .. 9 mm clearing between the wheels, Tillig, BTTB and other modern rolling material feature around 10 mm clearing.
And the crossings and turnouts are modeled accordingly, resulting in preventing ROKAL material from running over those new crossings.

Ok, I have a CNC router - should be easy enough to remedy this fact...
But have a look:

[image]

I built a rig to fix the moveable parts of the crossing (it's a so called "Doppelkreuzungsweiche" which allows diagonal transit as well as following the curve on both sides) for milling.

In the zoomed image you see the milling marks at the curved guidances (red circles):

[image]

As you may realize, the milling marks are not continuous but scattered along the path.

This is why I think I have to probe the height along the projected milling path to follow the obviously existing height curvature when milling.

The problem is twofold (at least):
FIrst, I have to define the path along which OCP has to get the probes (assuming that the probed heights along this path can be mapped to the curve).
Second, I have to align the milling path to exactly follow the guidance when milling.

This second topic is another problem in itself.
I tried to solve it by overlaying a photo of the crossing to the construction in the CAD program, but naturally there are always distortions from the optical system, so I had to make assumptions about the radius of the milling path.
The trench between outer rail and inner guiding is about 1 mm wide and the probing has to occur in this trench.

Here it would be very helpful if OCP had a feature to probe sideways and reckon the X/Y path for both, the subsequent height probing and the milling itself, respectively.

For my first tries (not very successful, I already killed three of those crossings) I tried to align the crossing by moving the stylus along the designated milling path without touching the material, matching stylus and material optically, over and over again until I thought, it might fit... varying results. As already mentioned, I had no 100% result up til now :(
Know, there are 4 four curved pathes and four straight pathes to mill (two for each side, respectively).

In this respect the probing sideways feature would be very appreciated :)
I know, this is a very specific feature, but please think about it anyways.
Thanks!

Harald

martin2250, 2019-05-02:
Hi Harald,

I'm sorry but this is going to be a definitive no (at least for implementing this into OpenCNCPilot). This is way to specific of a use case to warrant putting this into OpenCNCPilot. I did entertain the thought of allowing height maps with arbitrary probe positions (instead of a fixed grid), but the math required for this was too hard for me at the time.

That said, grbl is absolutely capable of doing this, but you'll need to write some custom PC software to go along with it.
Probing sideways is as easy as sending "G38.2 X/Y <...>" (not sure if you can move more than one axis at a time for probing diagonally)

I'd suggest picking some programming language of your choice (I'd use python for this kind of job) and writing a script specifically for this. For a quick proof-of-concept, you can also probe a few points by hand and use these points as reference in your CAD software. You can copy the exact locations of the probe result from the "manual probing" tab.

Martin

deHarro, 2019-05-02:
Hi Martin,

I expected some reaction in this direction, but having not asked would be gone only half way ;)
Thanks for the explanation, anyway!

One question is not answered, still: Is OCP capable of probing - and afterwards following - an one-dimesional height path? (following in the sense of adjusting a given g-code accordingly)
Or is the mathematics bound to a grid of at least two rows?
Without this capability all other ideas are in vain.

Thanks!
Harald

martin2250, 2019-05-02:
I'm actually not sure what happens if you set the size to zero along one dimension, this will probably result in an error. The math works in 1D (at least along a straight line) as well, but I'm pretty sure this would require implementing some special cases which normally aren't needed.

deHarro, 2019-05-03:
I tested setting one coordinate to zero.
The resulting test pattern collapses and there remains nothing to probe for OCP :(

I already imagined to set the probe coordinates by means of jogging commands and then do the probing with macros, but with this findings there is no chance.

Harald

deHarro, 2019-05-09:
Hi @martin2250!
Sorry, I just want to push this topic again...

Concerning my question:
Is it possible in principle and what should be the contents of my macro to get the probe readings into the "linear map" of OCP I described obove?

Or perhaps can I use "manual probing" for that?
(The mouse over help of "save pos" tell something different)

Thank!
Harald

martin2250, 2019-05-10:
Hi Harald,

sorry for not getting back earlier. There is no way to use macros to populate points of a height map. You'd need to edit the hmap file by hand to enter the probe results:
- set up your workpiece
- generate a 2xN point height map that spans the part
- for every x coordinate in the height map, move the tool to that x and eyeball the right y coordinate where you want to probe
- probe at this point (G38.2...) and transfer the Z height to the two associated points in the hmap file

To generate the toolpath:
- at multiple points along the track (spaced equally along x)
-- move the tool to that x
-- move y so you can plunge next to the wall that you want to mill
-- drop the tool to a z height below the top edge of the wall
-- use "G38.2 Y-5 F20" (or similar) to probe the exact position of the wall along y
- transfer the coordinates to your CAD tool of choice and use them as reference to recreate the curve of the track
- export the toolpath
- apply the height map as described above

I hope I covered everything, I'm pretty tired right now 
Martin

deHarro, 2019-05-10:
Hi Martin,
what a pity - but thanks anyway for the quick run-through!
Sounds reasonably enough and should work.
If not, I'll come back... ;-)

Thanks!
Harald

closed 2019-05-10 by deHarro