martin2250/OpenCNCPilot #110: Limit Z is ignored while surface scan

salihonur, 2019-08-12:
Hi,

The machine does homing XYZ properly. Limit-Z is activated when the bit touches to the pcb surface (clean copper). But Z axis isn't pulled back while surface scanning despite the Limit-Z appears in red font on the screen. How can solve this problem?
$21 Hard Limit = 0;

Because otherwise homing isn't done successfully although all limit switches works fine. First it touches and goes a little back and touches again slowly. Same for all XYZ axis.

martin2250, 2019-08-12:
Hi salihonur,
you need to use the probe input (Arduino Pin A5), not the Z limit switch input, for probing.

salihonur, 2019-08-13:
Thanks very much Martin. It works properly.

closed 2019-08-13 by martin2250