martin2250/OpenCNCPilot #112: Very big values should be checked

deHarro, 2019-10-07:
Hi Martin,

first of all: That's no issue of OCP!
But perhaps you can help.
I recently ran into a problem which caused me some headache and time until I found the reason.

I did a small board with some connectors and text on it, no big thing.
The etching of the traces and text performed well, I then turned towards drilling the holes.

Since my EAGLE ULP always insists on some (unneccesary) tool changes, I use to edit the file manually and get rid of those commands.

On the editing job I seemingly had done a mistake and a period in one number vanished.
Instead of G0 Z20.000 the line now reads G0 Z20000.

When GRBL encounters this big number it stalls and refuses any further cooperation, I had to reset GRBL (by means of toggling the DTR line or even restarting the PC) and restart this job from scratch (e.g. finding the original starting point in X and Y and zeroing Z).

Main problem was, I didn't hear the sound of my stepper motors, since I weared my noise protection. So I thought of EMI distortions, loose contacts and all sorts of electrical imponderables which can make a CNC router behave eratically. 
So I made many such restarts with noise protection on until I found culprit. This happened when I finally didn't start the vacuum cleaner and the spindle and just watched the movements, with ears unprotected.
I now heard that the stepper starts acceleration but gets interrupted and stops immediately without correct deceleration. This leads me to check the input to GRBL.

Here my idea comes in:
Perhaps you can implement some rudimental checks against implausible numbers into your already available checkrun when loading a job file?

Just an idea :)
I don't know how big numbers GRBL can deal with, but obviously it accepts numbers it definitely can't handle.

Harald

martin2250, 2019-10-07:
Hi Harald,

implementing such loose bounds checks wouldn't be that helpful. If at all, I'd implement it properly with the maximum axis ranges. Doing that would be pretty involved though, because of work offsets.  Also grbl already does a pretty good job at it.

After sending a command that exceeds the machine bounds, you should be able to reset grbl with soft reset (at least I remember doing that the last time I ran into my machine's limits). Usually you shouldn't experience any lost steps, but you should probably still re-home your machine afterwards. If you used G10 to set your WCS, you won't have to find your zero point again afterwards.

And of course you should always check the toolpath preview for such mistakes :)

Martin

deHarro, 2019-10-07:
Hi Martin,
I (some sort of...) checked the preview and I remember having seen some lines going out of the monitors area, but I gave no attention to that ("some lines" because I had two of those missing points. I did search&replace all).

Trust me, no option inside OCP gave me access to the GRBL controller, I tried all and some more ;-)
Power Cycle to the Arduino, restarting the PC and issuing a toggle of DTR (I think it was DTR, I cannot look it up, the program runs on my router PC down in the cellar and I'm upstairs... ;-) inside a terminal program were the only means of doing the job.

Ok, I have to confess, the X and Y coordinates were only half of a step away from zero, so I didn't had to adjust them, but nevertheless it was annoying.

GRBL was not offended by this big numer, it just crashed (or stopped?). At least it doesn't display an error message and wasn't responding afterwards.
Perhaps because I made no settings about the machine limits?
... No, I really adjusted $130..$132 to reflect my machines dimensions.

Must be some other reasons.

Ok, I think I will never run into this kind of problem, so don't feel pushed into doing this sort of check.
As said, just an idea ;-)
Cheers! 
Harald

closed 2019-10-07 by deHarro