martin2250/OpenCNCPilot #113: Spindle not running

mysecondnameis, 2019-11-19:
I often get problems when running my gcode, the spindle doesn't work and when I run a manual command with M3 it still fails.
im generate gcode using PCBGCODE 3.6.2.4

canbaytok, 2019-11-19:
Hi there,

grbl needs a spindle speed, so it knows how fast the spindle should be spinning. the command "S1000" followed by "M3" should start the spindle and make it rotate at full speed (if 1000RPM is the set max speed).

give it a try!

beckmx, 2019-11-26:
PCBGCODE generatesthe spindle speed in the section of feedrates should be at least 10000, if you havent placed anything there, you should run this macro:

```
S10000
M3
M2
```

That should run it, and if it doesnt run then is your version of grbl, btw, this issue is not related to opencncpilot, and shoulnt be raised here

closed 2020-03-19 by martin2250