martin2250/OpenCNCPilot #115: Pan the view

luotope, 2019-12-20:
Hello and thanks for the great software. 
My question may be too easy but I haven't found any instructions or manual on controlling the view on screen and cannot figure out how to pan the view to right or left. I have only touchpad and can zoom in and out and rotate with the right mouse button. But how to pan? The zero point is always at the center of the screen but if I want to move it to the right of the screen how do I do that? And sorry for the simple question.

canbaytok, 2019-12-20:
Hi there luotope!
You can pan the view by holding the middle mouse  button and moving the mouse around. Unless your laptop has a trackpad with a left, middle and right mouse button, you need to plug in a seperate mouse and use that for the viewing controls.

luotope, 2019-12-20:
Thanks for your response canbaytok. The problem I had was really that in my laptop there is only two buttons with a touch pad. But as I tried to edit and comment above, there is a solution for it: you need to use your right mouse button and use the keyboard left/right buttons with shift key to pan. It was a bit difficult to guess, but no need for an external mouse for it :)

closed 2019-12-20 by luotope