martin2250/OpenCNCPilot #116: simulation

lianzaozi, 2020-01-12:
Thanks for providing such excellent software. How can OpenCNCPilot software perform trajectory simulation without hardware connection? I looked for it in software and didn't find where this function is?

martin2250, 2020-01-12:
Hi @lianzaozi thanks, glad you like OCP. Currently, trajectory simulation isn't possible and I don't think that it will make it into the list of features. If you want to see the trajectory interpreted by grbl, you could connect a 'dummy' board with grbl or maybe play around with the grbl simulator (didn't work especially well when I tried it though).
Other than that, in the past I've used https://ncviewer.com/, which works pretty well and doesn't require installation.

lianzaozi, 2020-01-16:
I see, thank you for your reply

closed 2020-01-18 by martin2250