martin2250/OpenCNCPilot #121: Viewport handling when zooming

deHarro, 2020-02-26:
Hi Martin,

how about changing the handling of zoom actions?
At the moment zooming in OCP always refers to the center of the viewport.

So if you want to zoom in on a specific point it is a twofold process: Center this point and zoom

Other CAD tools (Eagle, DesignSpark mechanical, Estlcam to name a few) take the actual mouse coordiantes as anchor for the zoom process.
If you want to zoom in on a specific point you aim with the mouse somewhat outside of this target point and scroll the mouse wheel: The target gets zoomed in and simultaneously travels to the center of the screen.
I consider this behavior more intuitive.

Just an idea... 

Harald

martin2250, 2020-02-29:
Hi Harald,

in the other tools you've mentioned, does this only work in 2D views or also in 3D? I'd imagine this is a bit harder to do in 3D but I can look into it some time in the future. Thanks for the idea!

Cheers!

deHarro, 2020-02-29:
Hi Martin,
in Estlcam and Eagle there is no 3D Option, so no, only in 2D.
In DesignSpark mechanical the zooming works in both views, 2D and 3D, respectively.

But I think, the zooming works only on the magnification of the viewport, so it's no matter of dimensionality. At least this is the behavior of DS mech, as far as I can judge it.

Harald

terjeio, 2020-03-09:
I do it like this in my code:

`<helix:HelixViewport3D x:Name="viewport" Panel.ZIndex="0" ZoomExtentsWhenLoaded="False" RotateAroundMouseDownPoint="true" ZoomAroundMouseDownPoint="true">`

martin2250, 2020-03-10:
Thanks for the hint, I'll try that!

martin2250, 2020-03-19:
Thanks! It does behave a bit funny as it will only zoom to the location when the cursor is above a grid line.

closed 2020-03-19 by martin2250

deHarro, 2020-03-20:
Hi Martin,

according to [this post](https://github.com/helix-toolkit/helix-toolkit/pull/290) they fixed this behavior...

How can I test the new feature?
Wait for the next release? Or download the current development version?
Thanks anyway!!

Harald

martin2250, 2020-03-21:
Wow, I just noticed that I was using an old version of Helix Toolkit. Thanks for the hint! I've updated to the latest version, but I can't tell a difference in the zooming behavior. Anyways, I think this justifies a new release.

Martin

deHarro, 2020-03-21:
Fine. Please keep me posted...
Thanks!

martin2250, 2020-03-21:
Here you go: https://github.com/martin2250/OpenCNCPilot/releases/tag/v1.5.7

deHarro, 2020-03-21:
Hi Martin,
as you wrote: One has to set the cursor onto one of those fine lines (or at one of the designes lines? Didn't check that so far) to make zooming occur at the mouse position.
They discussed this topic in another thread over at Helix forum and said something about changing or defining or ignoring the Z coordinate to behave like expecte (sorry, I'm not so deep into coding C#).
Here is the [other thread](https://github.com/helix-toolkit/helix-toolkit/issues/289), perhaps you get what they mean...

Harald

deHarro, 2020-03-21:
Hi Martin,

checking done :+1: 

If I load a model (anything to etch) then I don't have to target exactly onto a single line, it's enough to set the mouse within the green rectangle you draw around the desing.
Please test it and report, whether I'm chasing white mice.. :)
Thanks!

Harald

deHarro, 2020-03-21:
Hi Martin,
sorry for writing again and again...

As stated, I'm no C# programmer.
But if I remember correctly, over at Helix they wrote something about getting rid of something like "find nearest" and changing to "onproject"... looked it up:
"'289 ZoomAroundMouseDownPoint calculation changed from viewport.FindNearest to viewort.unProject".

In your code I didn't find any of those words, but I realized that, when zooming with a project loaded, the picture zooms in and out somewhat jerky. The picture jumps around the mouse position in small increments during zooming (in or out, without moving the mouse).

This behavior leads me to think about FindNearest.
Is this perhaps the default handling? (Since I didn't find it explicitely in the code)
Why?
When a project is loaded, one can aim the mouse anywhere (within the borders of the project) and zooming occurs at the mouse coordinates.
Think of given the starting point of the zoom action is somewhere near, but not exactly ontop of a  grid line.
When zoomin occurs, it might happen, that the distance to the nearest gridline changes from left to right (or any other direction), so that the zooming center snaps to that other grid line, making the picture jump.

So: Perhaps you can change the viewport handling to this "unProject" thing and all is well...

I hope you get the point of my drivel :-)

Harald

martin2250, 2020-03-21:
Thanks for the investigation, I'll look into it!

martin2250, 2020-03-22:
Hi Harald,

the code you mentioned (Unproject()...) is contained in the latest version of HelixToolkit, so this isn't it. That said, stupid problems call for stupid solutions... I added an invisible plane to the scene, which  is targeted by the mouse when zooming. Now zooming works much better!

Martin

deHarro, 2020-03-22:
Hi MArtin,

zooming in 1.5.8.0 works like a charme :-) :-)
Thank you very much for your efforts!

Harald

commit 0e8af6ad (2020-03-19): make zoom follow mouse

close #121