martin2250/OpenCNCPilot #122: Marlin 1.1.9

tizi24, 2020-03-01:
Hi, I find this software fabulous and very interesting to adapt for Marlin 1.1.9 so that we can pcb with our 3D printers

tizi24, 2020-03-01:
I started to investigate a bit, you can use the "Marlin" G30 code to sense each point of the generated OpenCNCPilot mesh and then apply the different heights of the pcb

martin2250, 2020-03-21:
Hi @tizi24 

supporting multiple controllers right now would require a lot of changes in the code. Anyways I don't really see the need for this, as pretty much every controller intended for milling uses GRBL (or at least GRBL can be compiled to run on it). If you want to use OpenCNCPilot for your 3D printer, you have two options:
* change the relevant code and maintain a separate fork of this project (not much work, mostly regexes and other string parsing)
* (temporarily) flash GRBL onto your controller

That said, you still need to make sure your 3D printer is rigid enough, mine sure isn't. Do you already have a printer that's modified for milling?

closed 2020-03-31 by martin2250