martin2250/OpenCNCPilot #125: Pass through “unknown gcode” mode?

tarmon01, 2020-03-25:
It would be very nice if there was a “unknown gcode” pass through mode. Perhaps add a configuration option to either run in the existing mode or this new mode?

Thanks,

-TomH-

martin2250, 2020-03-28:
Hi Tom,

I've started to work on this, but I've hit some road blocks that make it too unsafe for most people to use yet.

I want a command like "G38.2 Z-10" to make the G-Code interpreter forget where it was in the Z-axis (the interpreter needs to know the end and **start** point of a motion and this leaves the machine in a state where the start position is not well defined). This makes it so you always have to send a "G0 Zxx" after the previous command to avoid errors. Right now, this G0 Zxx gets ignored if the position before G38 was already Zxx and making it not ignore this move requires a lot of rewriting.

I'll see when I can work on this some more. If you absolutely want to try it in the current state, clone the [passthrough](https://github.com/martin2250/OpenCNCPilot/tree/passthrough) branch and compile it with Visual Studio.

Martin

tarmon01, 2020-03-28:
Hi Martin,
I appreciate what you are doing! 
So yes I always do a G38.2 Zx when I change tools(work offsets in my case). Of course other people may not do that for normal use. Thus having a button in the config menu to switch modes from the "normal mode" to "pass though mode" would be a way to handle it.Perhaps put up a  warning window when you change modes to pass through?
Technically I don't use the G38.2 but instead use G38.3 as my probe is always connected to the tip and I have constant connection to the copper surface when a board is set to be made.
But it functionally it works the same. 
I have a "Center" macro in openCNCpilot that is a more elaborate "home" button that after it homed(in X and Y only)also sets up all the G10 offsets for the tools. This works as expected.
My Process:
I  generate the Gcode with the multiple tool numbers.
I then replace those tool numbers in a Python pre-processor script to generate inline Gcode tool change commands.
Here's a snip of the script.---for x in Inf:
  if x == ("T900"+'\n'):       ,< PCB bit tool number
    Outf.write("M5\n")
    Outf.write("( Carver )\n");
    Outf.write("G0Z20" +'\n')   
    Outf.write("G54" +'\n')
    Outf.write("M9" +'\n')
    Outf.write("G0X0Y0" +'\n')
    doprobe("Carver");
    Outf.write("M3" +'\n')   
    Outf.write("( Start )\n");
  else :
    if x == ("T910"+'\n'):  <-  Drill tool number
      Outf.write("M5\n")
      Outf.write("( Drill )\n")
      Outf.write("G0Z20" +'\n') 
      Outf.write("G56" +'\n')
      Outf.write("M9" +'\n')
      Outf.write("M8" +'\n')
      Outf.write("G0X0Y0" +'\n')
      doprobe("Drill");
      Outf.write("M3\n")
      Outf.write("( Start )\n");etc.---Where the doProbe routine is this
def doprobe(str):
  Outf.write("( set Z0 on "+ str+ ")\n");
  Outf.write("G49\n");
  Outf.write("G38.3 Z-33 F100\n");
  Outf.write("G10P0L20Z0\n");
  Outf.write("G92 z0\n"); 
  Outf.write("G0z1\n");
  Outf.write("G38.3 Z-2 F10\n");
  Outf.write("G10P0L20Z0\n");
  Outf.write("G92 z0\n");
  Outf.write("G90\n");
  Outf.write("G0Z2\n");
  return;
-----Basically the main problem is that openCNCpilot deletes a bunch of the code in the above like the G38 and G54-56, and the G10lines gets partially deleted. :(
When I get a chance I will look at the passthrough branch and look how to get Visual Studio going. This is just one of several projects I am working on, but it's an important one. 
Again thanks,
   -TomH-

tarmon01, 2020-03-28:
Oh.. I reread your post.. and I may not have responded correctly to what you were saying. 

You are saying that the before and after Z position ( with the g38 in the middle) has to be different for it to work correctly?

Bad
Z2
G38.2 z-10 
Z2 

Okay
Z2
G38.2 z-10
Z3

I can make that work. 

Thanks,

-TomH-

martin2250, 2020-03-28:
> You are saying that the before and after Z position ( with the g38 in the middle) has to be different for it to work correctly?

exactly! you can enter a list of codes to be passed through in the settings

tarmon01, 2020-03-28:
Nice! 

Thanks Martin! 

-TomH-