martin2250/OpenCNCPilot #127: Make path following file when starting OCP via double click

deHarro, 2020-03-27:
Hi Martin,

just an idea.

OCP now is able to open an associated file by just double clicking on the file. Very fine :-)

My usual way of working with OCP is, I start OCP by double click, do some work with this very file (e.g. etching the traces), then want to open associated files (e.g. drill file and milling file) residign in the same directory.
So I click on "file" "open" and... see the directory I used last year in november... :-(

I would gladly accept if OCP sets the current working path to the file opened via double click ;-)
But that said, it's is only an idea, no must have.

Harald

tarmon01, 2020-03-27:
That does sound pretty good to me.

-TomH-

martin2250, 2020-03-28:
Great addition, thanks!

closed 2020-03-28 by martin2250

deHarro, 2020-03-28:
No big thing, sooner or later you would have found it by youself... ;-)

commit 5743cab5 (2020-03-28): change initial directory of open file dialog to last file

close #127