martin2250/OpenCNCPilot #129: Resume after Pause behaves irrational

deHarro, 2020-03-29:
Hi Martin,

yesterday I tried something new, this time on my CNC router (not at the office PC).

I make the front of another WordClock. The characters are milled from the front side, then filled with resin. Afterwards the plate is turned around and from the back bins are milled to let the light of the LED through.

Ok. I'm on the backside, milling the first bin.
The bin has to be aligned exactly behind the character, I wanted this to check and clicked "Pause" in the file tab.
Then I used jogging commands to lift the spindle in Z, so I'm able to look into the bin.
The spindle now is several centimeters above the workpiece.
All is fine, I clicked on "File->Start", the spindle lowers to the new position and the job went on.

Curiously as I am, I tried that one more time, this time I sent the spindle not only away from the workpiece but also some distance away in X. Much better to look into the bins... :-)

Ok, all fine, "File->Start"...
Guess what happens...
I have to add, the job halted some mm deep into the workpiece when I clicked "Pause".
The spindle lowers to the new (or old?) Z height (remember, some mm depp into the workpiece) and then moves to the old X position, leaving a fine trench. Luckily this happens at a place where I can remedy it without much effort.

This said, do you think, this behavior comes from GRBL or from OCP?
In my mind it would always be safer to first restore X/Y and afterwards restore Z. At least on a router(?)

Thanks!
Harald

martin2250, 2020-03-29:
Hi Harald,

sorry that happend! The pause and resume functions in OpenCNCPilot really do just that. They interrupt the file and resume afterwards. If you move the machine during the pause, it will have the same effect as adding these movements to random places in the file. Adding a function to remember the last position would be a lot of work (the way it's currently programmed doesn't easily allow that) and could also lead to errors if it behaves differently from the way the user expects, so I don't think I'll change the current behavior.

Martin

deHarro, 2020-03-29:
That's ok for me, I think (and hope) I remember that, if I use Pause in the future... ;-)

But just to get it right:
The gcode has absolute coordinates, so there is no need to "remember the last position".

Is GRBL or OCP responsible for the sequence of resuming moves?
Someone (GRBL or OCP) has to decide which axis is moved first.

Additional info:
As far as I remember the described situation, the bit first dug into the board, then linearly moved to the next absolute XY coordinate in the file.
In contrast to that, if you define a XYZ coordinate like G1 X10 Y10 Z10 f10, the bit moves linearly to the given point, all three axes at a time.

So in fact there is a different behavior in both cases.
The Difference:
In the file there is one command defining the Z depth for all successive moves in XY.
The resume (File->Start) tells GRBL what? 
"Continue where you left off"? Presumably not. OCP only passes the new coordinates in XY and GRBL knows "Z is different from the last gcode command, so let's  restore Z first".

I remenber having read in the GRBL documentaion that there are two totally distinct buffers for gcode streaming and jog commands. I will reread that (hoping that I will find it :).

But up til now I think, the described failure is GRBL's one.

Harald

martin2250, 2020-03-31:
Hi Harald,

grbl doesn't "restore" the Z height. grbl just executes commands in the same order that they appear in the file, same for OCP. Your mishap with the bit plunging into your workpiece was the result of pausing the file at an unfortunate moment.

The two buffers for streaming and jogging don't have anything to do with that, this is only a convenience thing where grbl accepts commands with the prefix "$J=" and doesn't go into an error state when the command exceeds the machine limits.

Martin

closed 2020-03-31 by martin2250

deHarro, 2020-03-31:
Hi Martin,

just to get it sorted in my mind...

I think I'm right whith my opinion, that GRBL in deed does a restore on Z.
Why?
Don't know it exactly, but there is something like "modal" commands and I understood(?) that a command like G1Z-0.5 will define the Z height _for all successive commmands_ issueing only X/Y movements.
Given that, and the fact that I paused the gcode at a location where only X/Y movements are streamed (the next command in the file was G01 X172.5195, no Z value), it is clear for me, that GRBL does a restore of Z.
Who else should have made the router go to Z-0.5?

This is the part of the gcode I'm referring to (I marked the line where I paused):

`(Nr. 2: Ausschnitt 4)
G00 X173.9745 Y13.6145 Z5.0000
G00 Z-2.5000
G01 Z-11.0000 F200
G01 Y16.0110 F800
G01 X176.3995
G01 Y13.6145
G01 X173.9745
G01 X172.5195
G01 Y17.4660
G01 X177.8546
G01 Y12.1595
G01 X172.5195
G01 Y13.6145
G00 Z5.0000
G00 Z-10.5000
G01 Z-11.0000 F200
G01 X171.7650 F800
G01 Y18.2350
G01 X178.5950
G01 Y11.4050
G01 X171.7650
G01 Y13.6145
G00 Z5.0000
G00 X173.9745
G00 Z-10.5000
G01 Z-17.5000 F200
G01 Y16.0110 F800
G01 X176.3995
G01 Y13.6145
G01 X173.9745 
G01 X172.5195            <<<<<<<< Pause
G01 Y17.4660
G01 X177.8546
G01 Y12.1595
G01 X172.5195
G01 Y13.6145
G00 Z5.0000
G00 Z-17.0000
G01 Z-17.5000 F200
G01 X171.7650 F800
G01 Y18.2350
G01 X178.5950
G01 Y11.4050
G01 X171.7650
G01 Y13.6145
G00 Z5.0000

(Nr. 3: Teil 18)`

I actually paused somewhere else before and GRBL emptied its buffer before it stopped.

Harald

martin2250, 2020-04-01:
Hi Harald,

grbl doesn't keep a state in that sense. grbl only knows its current position and nothing else. Also there is absolute nothing special about the Z-axis, grbl treats all axes equally. So going by your reasoning, X and Y would have to be restored as well.

The plunging motion is just a regular command from your file (maybe `G00 Z-17.0000` for example), which was executed when the machine was not in the state expected by the file. I don't know how to prove it to you, maybe you could do some experimenting without a workpiece if you doubt it.

Martin

deHarro, 2020-05-02:
Hi Martin,

I posted my question [over at GRBL ](https://github.com/gnea/grbl/issues/843#issue-589807684) and now they [answered](https://github.com/gnea/grbl/issues/843#issuecomment-622574229).
I don't know whether the author is an insider to GRBL.

In short:
On resume (from pause and jogging) GRBL does something, but doesn't know what...

The author of the answer claims, that OCP has to prohibid jogging when a job is paused.

In this context:
Is it allowed to jog when a job is paused by a tool change command?
Are there other occasions where jogging might be allowable during a (paused) running job?

Harald

martin2250, 2020-05-02:
Hi Harald,

I stand by my statement: grbl doesn't keep a state in that sense. It simply executes commands in the order that it receives them.

Of course this means that you can mess up a part when you move your machine while a file is paused. There is no way, that is 100% safe, to account for the user moving the machine in the middle of a job and simply hitting 'continue' (not even retracting to a safe height, image using an undercutter or thread mill). This is why OpenCNCPilot doesn't restore the position when a file is resumed, it leaves this responsibility to the user.

> Is it allowed to jog when a job is paused by a tool change command?

Usually a too change command is followed by some code that moves the machine to an absolute position in a safe way, so in most cases yes.

> Are there other occasions where jogging might be allowable during a (paused) running job?

Yes, it's completely fine to jog your machine while a file is paused. As long as you know what you're doing. Simply look in the file tab, there you can see which commands will be executed next. You only have to restore the machine's position in a way that won't result in a crash when the next commands are executed.
I don't think prohibiting the user from issuing commands is a good solution either. I often found myself pausing a file to lower the WCS a bit (when traces were not cut deep enough).

Martin

deHarro, 2020-05-02:
Hi Martin,
I'm with you, that seems to not come over, sorry.

> You only have to restore the machine's position 

Is there a way how I can save the current position (by a macro) into some GRBL variables?
And do a restore on X/Y and afterwards on Z by another macro?

Perhaps this can be established with 38.x commands?

I have to admit, that I sometimes do not understand the descriptions that come with the g-code commands :(

Harald