martin2250/OpenCNCPilot #130: Change arrow buttons for manual  jogging?

tarmon01, 2020-03-29:
Martin,

My x and y axis are swapped from your defaults.
Which is slightly annoying when I do manual jogging.

Is it possible to program which arrow keys do what ? Or at least be able to swap the X and Y axis?

Thanks,
 -TomH-

eried, 2020-03-30:
This is a setting in GRBL, have you checked something like: "$2 – Step port invert, mask" on https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration ?

tarmon01, 2020-03-30:
Thanks. Yeah. I could  do that. The axis are going the way I want for carving right now.  It’s just the jogging via the arrow keys that would be nice to map differently. 

If Martin doesn’t want to do it, I fully understand and can cope.

Thanks,

-TomH-

eried, 2020-03-30:
But how are they going the right direction? I mean the jogging direction is the same as the gcode in my setup. It would be highly confusing otherwise

tarmon01, 2020-03-30:
Take your keyboard( notebook in mycase)  and and rotate it 90 degree in orientation to your machine.   Pushing the arrow key that is in the same direction as the machine direction you want it to go is now the wrong axis.  As I said, it’s a minor annoyance. I don’t have space to rotate the machine or computer which would also fix it.  A arrow key mapping would do it. But then again I can cope. It’s more of a wish than anything else.

-TomH-

eried, 2020-03-30:
Well, maybe it is a good idea. I am used to shift+up/down for Z, so I would also use that "remapping" functionality.

martin2250, 2020-03-31:
Hi Tom, Erwin,

adding the option for completely custom key mappings (also for macros) has been on my 'nice to have' list, but right now I don't have enough time to check items off that list.

Shift+up/down should work though, that's what I always use on my laptop without dedicated page up/down keys.

Martin

tarmon01, 2020-03-31:
Thanks for considering it. 

-TomH-

eried, 2020-03-31:
> (also for macros)

For the macros is a bit more work but for jogging I could add something in Settings in this corona times and do a pull request :P

martin2250, 2020-04-01:
That would be great! A simple ComboBox with all four directions will do the job just fine I think (jogging -> "up arrow moves' -> '+y', '+x', etc.).

Thanks Erwin!

tarmon01, 2020-04-01:
Excellent!

Thanks,

-TomH-