martin2250/OpenCNCPilot #131: Custom mesh

eried, 2020-03-30:
How can I do manual probing for doing mesh corrections? Like jogging and then choosing a place I like for a new mesh probe point. I only see options for the tool length in the "manual probe" section

martin2250, 2020-03-31:
This is currently not possible. I've already thought about doing something similar for avoiding holes that were already drilled, but it would make the math more complicated. Right now it's simple bilinear interpolation and arbitrary meshes would require triangulation, which I knew nothing about when writing this.

eried, 2020-03-31:
Yes, I also wanted to avoid some holes and clamps. Maybe an easier solution is to add "probing controls" that allows you to skip some probe spots (interpolating them)

martin2250, 2020-03-31:
That could work, though skipping a point would leave quite a large gap (twice the grid size). I'll explore a few ideas on how to create the UI for this (maybe add a checkbox to the probing tab to overlay checkboxes over the 3D viewport). Not sure when I'll get around to it though.

tarmon01, 2020-03-31:
If you skipped a point, you would fill in that point’s data with an average of its closest neighbors that had real height data. Correct? 

Just wondering.

-TomH-

martin2250, 2020-04-01:
Not sure yet how I'd implement it. I'll have to give that some thought when I tackle this, maybe I'll also leave up to the user to choose how it's handled.