martin2250/OpenCNCPilot #132: Save/Restore Viewport rotates one time after saving

deHarro, 2020-04-10:
Hi Martin,
just encountered another neat feature of OCP :)

Set the viewport orientation to your liking.
-> Move Viewport somewhat by Middle Mouse Key (Wheel)
Save this setting.
Restore this setting -> Viewport rotates by 180°
...
Mhmm, occured three times in series.
Now, just to be sure I didn't forget a step, I tested again and there is no problem.
Always restores as saved before... 8-(

Ok, then.. Happy Easter!
...
Tried some more options and heureka!
See list of operations above.
The move operation seems to have some cm displacement to get the effect I described.

Harald

martin2250, 2020-04-19:
Hi Harald,

it may be a bit late, but happy Easter to you too!

I just tried to reproduce this behavior and it occurred first try... then I went into Visual Studio to debug it, re-built the project and now I can't reproduce it anymore? I'll leave this issue open and I'd suggest you to try to reproduce it when the next version is released. Very strange indeed.

Thanks!
Martin

deHarro, 2020-04-19:
Ok, I'll test it with the new release.
It's not there already, isn't it?
Harald

martin2250, 2020-04-25:
Nope, still waiting for a few more changes to "accumulate".

deHarro, 2020-04-25:
Ok, kein Ding :-)

 

 

Gruß

Harald

_______________________

Modellbau&Elektronik

 <http://www.harald-sattler.de/> www.harald-sattler.de

martin2250, 2020-05-01:
so, jetzt aber :) Konnte das Verhalten auch im neuen build nicht reproduzieren.

Grüße
Martin

deHarro, 2020-05-02:
Hi Martin,
I'm so sorry to tell you... it still occurs.

I'm not able to get the initial sequence to make it happen, but if I got the 180° flipping, it occurs again and again.

Perhaps it has to do with a "Reset 3D Viewport" operation I issued during my tests.

First I did the following:
Start OCP 1.5.9.0 for the first time after copying all files to the old location.
Open a design
Rotate the design by RM to be roughly aligned horizontally
Save Viewport Orientation
Restore Viewport
Nothing special, all as expected.

But then I issued a "Restore 3D Viewport" inbetween:
Start OCP 
Open a design
Restore 3D Viewport
Rotate the design by RM to be roughly aligned horizontally
Save Viewport Orientation
Restore Viewport
-> here the design rotates by 180°

When I then rotate the design to any orientation (to realize whether it rotates again or not) and save and restore, it is rotated by 180° every time from then on.

Harald

martin2250, 2020-05-11:
Hi again, I'll give it another go when I have some spare time. For now I think we can still live with this bug :)

Cheers,
Martin

deHarro, 2020-05-11:
Fine, no hurry! :-)

deHarro, 2024-09-08:
Perhaps it's fixed with my last [update ](https://github.com/martin2250/OpenCNCPilot/pull/192)to "Lay flat..." (as of 08.09.0224)