martin2250/OpenCNCPilot #133: Additional variables in expressions label:"nice to have"

dilbert283, 2020-04-28:
Hi Martin,
Great job!

Could it be possible to have additional variables for use in expressions? I would be interested to have variables with the min/max coordinates of the current job (i.e the same values as displayed in the 3D window).

Usually when I start milling a new PCB, I first jog the machine to all 4 corners of that PCB to be sure that it fits in the remaining available material. With the mentioned variables, I would be able to have a macro that would automatically move to that 4 corners, so that I could quickly visually check that all will fit.

Bertrand

martin2250, 2020-04-29:
Hi Bertrand,

great idea! I'll try to implement this soon :)

Martin

martin2250, 2020-05-01:
done :)

closed 2020-05-01 by martin2250