martin2250/OpenCNCPilot #134: Unpinning and/or moving and/or resizing some of the GUI elements

deHarro, 2020-05-02:
Hi Martin,

someone (aubete?) rised this topic in the past, but you didn't comment on it, or I passed

On my office PC I have no concerns with the arrangement of the different sub windows of your GUI, the screen is big enough to show all information and buttons I want to use.
But on my routers PC the screen is smaller and I often have to close and reopen some of the subwindows to get to my macros.

In the first approach it would be helpful to just be able to move the macro window to another location (e.g. by changing the order of the available windows).

Silver bullet would be, to unpin all subwindows from the GUI and locate them freely over the screen, inside or outside the GUIs area :-)

Another goody would be to resize the width of the file control (to be able to read longer lines of the g-code).

Just in case you have some spare time and nothing else to do... ;-)

Harald

martin2250, 2020-05-11:
Hi Harald,

thanks for bringing this up again! Freely positioning all tabs would take a lot of work, but changing the order of tabs and making the file preview resizeable should be doable. Might be a while until I get around to it though.

Cheers!

deHarro, 2023-02-22:
Hi Martin,

just want to bring this topic to your attention again :)

Do you plan (and will have time) to work on enhancements at all?

Harald

martin2250, 2023-02-22:
Hi Harald,

I still live separated from my machine ( 😢 ), so I'm currently not using OpenCNCPilot and it doesn't get a lot of my attention. I'll definitely try to fix the other issue you opened when I get back from my vacation, then I can at least check if it is possible to implement at least some of the suggestions with little effort.

Martin

deHarro, 2023-02-22:
Hi Martin,

ok, i'll stay tuned :)
Have nice vacations!

Harald