martin2250/OpenCNCPilot #135: Upgraded to OPC 1.5.9 and can't turn off the grid?

happycncer, 2020-06-01:
Had 1.5.6 up until recently and have now just upgraded to v 1.5.9. On 1.5.6 I had somehow managed to turn off the grid in the viewport somehow - cant remember how now, but it looked a lot cleaner (to me) whist doing small mundane things that didn't take a lot of effort to follow.

Went looking to make 1.5.9 the same and I cannot seem to find where and how I changed it - and I now don't have 1.5.6 to even look at as I deleted it straight after installing v1.5.9.

Can anyone please enlighten me as to exactly how I managed this before - cos I can't find it now!
happycncer

deHarro, 2020-06-01:
Set "grid thickness" to 0 (Zero) and your done :-)

[image]

Harald

happycncer, 2020-06-01:
Duh, I just knew it'd be something simple! (a bit like me?)

Many thanks Harald, appreciated....
happycncer

closed 2020-06-01 by happycncer