martin2250/OpenCNCPilot #137: Upgrade issues?

happycncer, 2020-07-13:
Have just upgraded OCP from 1.5.6 to v 1.5.9 and have noticed something strange...

In OCP v1.5.6 I used the following Macro to zero out the XYZ locations,
G92 X0
G92 Y0
G92 Z0
G91
G21
G38.2 Z-25 F75  
G92 Z11.85
G0 Z3
G0 X-25
G0 Z-10        
G38.2 X20 F75
G92 X-11.30
G0 X-3 
G0 Y-25
G0 X30  
G38.2 Y30 F75
G92 Y-11.40
G0 Y-3  
G0 Z10
G90 

This has always worked well for me. However, I thought it would be a good idea to add comments while upgrading and added them as shown:-

G92 X0			(Set Position to Value)		
G92 Y0			(Set Position to Value)
G92 Z0			(Set Position to Value)
G91			(Incremental Mode)
G21			(Set Units to MM)
G38.2 Z-25 F75		(Probe Z (Down) until Touch)  
G92 Z11.85		(Set Position of Z Axis to 11.85mm (Plate Thickness))
G0 Z3			(Rapid raise of Z Axis 3mm)
G0 X-25			(Rapid move in X Negative(Left) 25mm)
G0 Z-10			(Rapid move in Z Negative(Down) 10mm)        
G38.2 X20 F75		(Probe X Positive (Right) until Touch)
G92 X-11.30		(Set Position of X Axis to -11.3mm (Plate Margin of 10.3mm + 50% of bit Dia))
G0 X-3 			(Rapid pulloff Negative (Left) of X Axis 3mm)
G0 Y-25			(Rapid move in Y Negative(Toward Front) 25mm)
G0 X30			(Rapid move in Y Positve (Right) 30mm)  
G38.2 Y30 F75		(Probe Y Positive (Away from Front) until Touch)
G92 Y-11.40		(Set Position of Y Axis to -11.4mm (Plate Margin of 10.4mm + 50% of bit Dia))
G0 Y-3			(Rapid pulloff Negative (Towards Front) of Y Axis 3mm)
G0 Z10			(Raise Z 10mm to clear workpiece)
G90			(Set Absolute Positioning)

I'm now finding that when I run the annotated version under OCP v 1.5.9, the initial Z movement goes upwards and activates Z- Limit switch, rather than downwards as it should do. If I run the version without any annotation under v1.5.9, all works as it should?

I can't see for the life of me why one should act differently from the other as the basic code is identical in both Macros?

Am I missing something here, or is it just not a good idea to annotate the macros?

As ever, any help in answering this query would be greatly appreciated.....

deHarro, 2020-07-13:
Hi!

 >why one should act differently from the other

Did you made the reverse check?
Try to run your annotated macro under 1.5.6 and see what is does...

Harald

happycncer, 2020-07-13:
Hi Harald,

Have just made the reverse check and yes, the annotated Macro does the self same thing in OCP 1.5.6.

It sends the Z Axis toward the Limit Switch every time. Thought it might have something to do with the fact that I had used 'Tabbed' moves between the gcode instruction and the annotation to align the comments.  I removed all these tabbed spacings and tried again with 1.5.6 and the result was exactly the same - sends the Z axis up towards the Limit switch every time.

So, drawn a blank on that one I'm afraid. Any other ideas -(it's starting to bug me now!)....

deHarro, 2020-07-13:
I think you have to "export" your annotated version of the macro to some sort of notepad for reference nad delete the annotations in the macro to work with.

But you should wait for Martin to comment on this issue. He will know for sure whether one can commment macros or not.

happycncer, 2020-07-13:
OK - will do that. It just intrigues me as to why this should make the difference that it does?

Many thanks for your suggestions.........

martin2250, 2020-07-26:
I did some testing and it seems that this is an issue with grbl, not OpenCNCPilot.
The issue arises when you nest comments (use brackets inside brackets).
To reproduce this, send the following in the "manual" tab:
```
G0 Z10 (test)
G0 Z11 (test(test))
G0 Z10 ; test (test) (test(test))
```
the first line will work just fine, the second line will produce the same error as your macro and the third line will work just fine again --> If you want to add fancy comments with a lot of brackets, use semicolon-style comments.

silicon-systems, 2020-07-27:
> I did some testing and it seems that this is an issue with grbl, not OpenCNCPilot.
> The issue arises when you nest comments (use brackets inside brackets).
> To reproduce this, send the following in [...]

Hi Mr. Pittermann, but OCP frowns on such use of semicolon-style comments [;].
Hope I'm just wrong.

Hanspeter.

martin2250, 2020-07-27:
Hi Hanspeter,

you're absolutely right, I'll try to fix this soon!

Cheers,
Martin

silicon-systems, 2020-07-28:
> 
> 
> Hi Hanspeter,
> 
> you're absolutely right, I'll try to fix this soon!
> 
> Cheers,
> Martin

Thanks Boss.

OCP seem not to send 0x18, [ctrl-x] or [Alt+25] on numeric pad, for Soft-Reset.
[~ ! ?] these seem to work ok.
Although you have explicit button for this in OCP for this, I want to send a concoction from
a macros ;-)

Hanspeter.