martin2250/OpenCNCPilot #140: Runtime display gets reset when pausing/restarting job

deHarro, 2020-08-05:
Hi Martin!

Besides the fact that the runtime forecast is fairly unsecure (you mentioned that in the beginning and I accepted :), I recently found, that the passed time gets reset if I pause and resume a running job.

I think, getting this value correct will be a very hard job since the fact "paused" doesn't mean, that the router stops immediately. The already filled buffers will be emptied before, so even if you stop the timer, the job continues several seconds, depending on the complexness of the job. E.g., if I plane a big area with long linear moves, it can even last for minutes until the buffers are depleted and the router stops.

That said, I have no problem with the current implementation, but if you have a brilliant solution (besides completely killing this feature ;-) please fix it.

Harald

martin2250, 2020-08-06:
Hi Harald,

I just tested this and the run timer keeps on ticking until the CNC has stopped moving (Idle state) and the timer is not, as you stated in the title, reset. It is only reset when a new file is loaded.

Is there anything you did which I need to reproduce this?

Martin

deHarro, 2020-08-07:
Mhmm, I cannot reproduce it (at my office PC), either. 
I will try to get it at the router and report. Just can't remember, what I did back then, when I realized it the first time... if ever :-)
Harald

deHarro, 2020-12-11:
Hi MArtin,

didn't realize it in the meantime, I'll close this topic (for now ;-)
Harald

closed 2020-12-11 by deHarro