martin2250/OpenCNCPilot #143: "Apply HeightMap" should be possible only once -> Graying

deHarro, 2020-09-01:
Hi Martin,

I remember, that this topic was raised in the past by another fellow, but I didn't find it.
Perhaps it's part of one of the lengthy discussion threads and got missed...

Please implement graying the "Apply HeightMap" button when it is issued to the current file.

One for sure can work around for sure by reloading the file and then be attentive to one's own clicking around, but being guided by OCP would be fine, anyhow :-)

Harald

martin2250, 2020-09-25:
Hi Harald,

sorry, the email got buried. I'll work that into the next update!

Martin

martin2250, 2021-02-28:
Hi Harald,

I added a warning in 1125b642db618bd4ca441eabe7ceb4fa45e59187.

Grüße,
Martin

closed 2021-02-28 by martin2250