martin2250/OpenCNCPilot #145: Using Gcode from Flatcam

slorole, 2020-09-22:
For starter's I love this program.
It works great with gcode coming from pcb-gcode.  But coming from Flatcam its another story.
when you first load the gcode it has a long list of errors. maybe it just don't support files coming
from flatcam.
I have only found one other person with this issue and he had no answer.
I attached a .png so maybe you can see my issue.
But it's still a great program.  Thank You
[image]

martin2250, 2020-09-23:
Hi @slorole 

you can safely ignore all these warnings. I should probably get rid of these 'zero legnth' warnings.

The second warning is a bit more interesting, it means that the gcode uses G1, without setting the feed rate (Fxxx) first. You should try to find out why, but the code will probably still run fine.

Cheers!

slorole, 2020-09-25:
Well it seems to work just fine after some adjustments in Flatcam. 
I think it looks good on my scrap piece's anyway.
                              thank you for your Time and Work
[image]

martin2250, 2020-09-25:
the bottom PCB looks pretty clean!