martin2250/OpenCNCPilot #146: Something for the 'non-metallic' workers maybe?

happycncer, 2020-10-20:
Would like to make a request to see if it is at all possible to add some kind of macro or widget to help out the non-metalworking fraternity.

The primary materials I work with are plastics, thin plywoods, melamine and corian - all of which are, of course non metallic. Thiis means I have to use a workaround to get the best use from your fantastic program. What I do is to use a metallic tape for the probing routine in order that I can get all the benefits of autoleveling. This in turn means applying and removing the tape between creating the heightmap and actually cutting the project out (or engraving which is mostly what I do..). This is very time consuming - apart from being exceedingly boring!. There is also the chance of accidentally moving either the part or the machine in doing this removal - especially if it is a large part.

It occured to me that it would be a great idea if you could have a small piece of code/Macro/widget or whatever, that would allow me (and others) to use a 1 or 2mm plate which could be used to manually 'follow' the probe routine, and then be compensated for after the completion of the probe routine to adjust the Z height of the spindle - compensating for the thickness of the plate _prior to running the actual program_. Maybe in the same way we currently apply the heightmap to the gCcode?

This would be a great advantage for me personally - but I'm sure it would also help a whole load of other people who, like me would like to use your great software all the time. A lot of woodworkers also have this very problem and would (I think) jump at the chance of using such a great piece of software.

deHarro, 2020-10-20:
Hi again!
I addressed this some time ago and [modified a 3DTouch sensor ](http://www.harald-sattler.de/html/3dtouch_sensor.htm)so that it can be used on our routers, controlles by GRBL V1.1 and OpenCNCPilot.
Some time later I [did it again ](http://www.harald-sattler.de/html/3dtouch_sensor_v2.htm)and found, there should be an easier way to get those sensors cooperate with GRBL.
The work on this [easier solution ](http://www.harald-sattler.de/html/3dtouch_interface.htm)is underway, but I didn't find the time to finish it.
The hardware seems to be already working, but the software has still to be crafted.

nota bene:
All my pages are in german, but at the very end of each page you can fire google translate.
But be warned, the google translations are sometimes not easy to understand. But in the given context the reader should be able to derive the sense of some crude sentences, google produces ;-)

Harald

firetunes, 2020-10-20:
@happycncer I carve exclusively on wood. I created my own dirt cheap probe that works very well. It is a momentary switch glued onto the end of a bolt. Simple, but very effective.

Here is a video about it:
https://photos.app.goo.gl/nDD9VguQ92NxMtMX7

You would have to re-zero your tool after generating the height map, but that shouldn't matter.

Cheers,
Ethan

happycncer, 2020-10-21:
Hi Ethan (firetunes),
I have tried something very similar before, but I always had shedloads of trouble trying to get the z height correct after the mapping has completed. Generally, I had to remove the switch from the spindle collet and replace it with the cutting bit (which of course was a different length from the switch unit) and then try to re-establish the correct X,Y and Z coordinates in relation to the heightmap.

This is where I always seem to run into trouble, I can _never_ seem to get it right. It would be great if you could explain _exactly_ how you manage to do that with your setup. How do you reset the Zero after completing (and saving) the height map?

Taking it just little a bit further, let's suppose I have a heightmap saved and shut the machine down for the night, using your system - can I go back the next day and load the saved map and set Zero as if I had just carried on the previous night?

Your comment 'You would have to re-zero your tool after generating the height map, but that shouldn't matter.' sounds oh so simple, but exactly where do you have to set that Zero to?

I've always found that if I don't go straight from the completing of the heightmap into the running of the Gcode program, I wind up having projects cutting far too deep or even cutting fresh air! I've lost count of the amount of jobs I've had to cut two (or three) times before getting something acceptable!

I'm sure there must be a way of doing it - it's just I haven't stumbled across the right answer yet. It's a problem that I've lived with for a year or more and usually, I normally end up being really anal about mounting my stock and getting it as level as I can. It usually turns out OK - but you always look at the job afterwards and think - maybe I could have done a better job.

Many thanks for your thoughts
Richard

firetunes, 2020-10-21:
Hi Richard, ( @happycncer  ),

It's a pretty low quality production, but here are some videos I made of the process:

https://photos.app.goo.gl/GrPV5ydx7Y6bMpQj8

Here is my process step by step:

1) With v-bit installed, locate X and Y coordinates relative to my workpiece. Set this as my new 0 position.
2) Raise spindle, replace cutter with probe
3) Run custom macro to Zero the probe (moves to X7.5 Y7.5 then zeros the probe)
4) Run probe routine to generate height map
5) Modify gcode w/ new height map
6) Raise bit and replace with cutter
7) Move to X7.5 Y7.5
8) Zero Cutter to work surface (I do this with a feeler gauge then step down until it's touching) (Set Z0 here)
9) Move back to X0 Y0 Z5
10) Run gcode (Note, I'm using Easel as my sender, Haven't fully switched to OpenCNCPilot)

Hope this helps!

Cheers,
Ethan

happycncer, 2020-10-21:
Ethan,

Many thanks for your detailed explanation - most helpful. I take it that X7.5 Y7.5 is the centre of your (cutting bed or) workpiece?

Also think there maybe  a 5a missing to save the modified gcode?
And maybe a 5b to shutdown OpenCNCPilot and open Easel - unless you can run both at the same time?

On step 9, do you raise the Z to Z5 because that is where the saved modified gcode file starts from? 

And finally, is there any reason that you don't use OpenCNCPilot as your sender?

Richard

firetunes, 2020-10-21:
Hi Richard,

The X7.5 Y7.5 is just to move the cutter into the workpiece as I set the X0 Y0 right on the lower left corner of the work piece. This way when I zero the actual tool to the work piece I do it at the same location as when I set the zero with the probe. Just for consistency.

And yes you are correct about 5a and 5b.

I raise the bit just to have it off the workpiece. No real reason.

There are some tools i prefer in easel at the moment over OpenCNCPilot, but I'm working on my own fork of the program and am working to add those tools into the program.

Cheers,
Ethan

happycncer, 2020-10-21:
Thanks again Ethan - glad I was reading it (almost) right.

You've been a great help, and good luck with creating your own fork of openCNCPilot.....

Would like to try it when you have a release - keep me in mind.

Richard

happycncer, 2020-11-18:
Ethan, I have tried your system of working now an I still cannot make my machine do what yours seems to do!

Having looked all through your videos yet again, I have now combined all your snippets into one continuous video, I have noticed that you use Easel to create your gcode as well as running your CNC Machine to actually cut out the actual piece.

However, I am using Cut2D to create my gcode and OpenCNCPilot to apply the heightmap __and_ then_ send the resultant codes to my shapeoko 2. What I am finding is that it seems to put things about 5mm too deep into the material I'm cutting into - every time! I can't seem to work out why this is happening - gcode programming not being a strong point with me. This is all very frustrating for me because when I do a conventional autolevel and cut on a PCB for example, everything works fine - presumably because I move seamlessly from applying the heightmap into the actual cutting sequence. I can't do this when probing a non-conductive surface

It's when I use a Z probing switch in the spindle bit holder to create a heightmap and subsequently swap-out the switch for the cutter bit, it always seems to go wrong. Hence my request to you for the exact way you do this. 

I'm sure it has something to do with the fact that I have to effectively disconnect the usb connection while I change the switch for the cutter - even though I do reset the machine to the correct position having completed the change of bits, it still always goes wrong by what appears to be, he same amount each time (5mm?)

Any ideas as to why this should be happening?

Richard

closed 2020-11-18 by happycncer

martin2250, 2020-12-11:
Hi Richard,

why do you "effectively disconnect the usb connection"? If you use OpenCNCPilot for both probing and cutting, you can leave everything connected. After changing the probing tip to the cutter, you only need to re-zero your z-axis (for example using G92) and you should be good to go.

Martin

happycncer, 2020-12-11:
I actually have to disconnect as I cannot swap-out the switch for the cutter bit as there is not enough height available to me on my Shapeoko 2 Z axis - to be able to remove the longish shank I have attached to the microswitch. I have to move the axis beyond the outside of my normal work area (beyond my normal soft limits) so that I can lower the shank out from the ER11 chuck. A bit like you see people doing on the shapeoko 3 - where the x axis actually overhangs the front of the wasteboard nearest the operator.  Shame I can't do the same on the older model!

I have the normal 18mm wasteboard on my shapeoko 2, plus a second piece of 18mm as a strengthener/sacrificial working wasteboard, _plus_ the actual workpiece I am trying to Carve/Cut/Engrave. As a result, I frequently seem to run out of height and have to use the additional space (down each side of the Y axis) which is outside of my normal work area.

Suppose the ideal would to be to raise the whole machine to give me more height - but  I'm trying to avoid that as it is a lot of hassle doing that! Just thought I would see if anyone had other methods of doing what should be fairly simple?

Another option that has only just occurred to me is I could shorten my sacrificial board at the front of the machine. This would then give me an area within my work envelope which is some 18mm lower where I could do a swap-out of the 'switch to cutter bit' similar to that use on the Shapeoko 3. 

Wow, I think maybe I've just talked myself into answering my own problem!

deHarro, 2020-12-11:
Hi Richard,
you alternatively (or additionally) may shorten the shank which holds your switch in the ER11 chuck. There will - or at least should - be no load sideways (in Y/Y direction) during probing, so a few mm insertion will be enough to hold the switch in place.

Harald

happycncer, 2020-12-11:
Hi Harald,

Already decided (since my earlier thoughts) to definitely shorten my sacrificial board and I shall now certainly take up your suggestion and take a grinder to the shank on my depth gauge/switch - just to make the whole thing that little bit easier...
Many thanks, Richard