martin2250/OpenCNCPilot #147: Retraction Height when Probing

happycncer, 2020-10-31:
OK Guys, worked it out now. Have to say language used is not that intuitive, I think this particular title would be better named as 'Retraction Height' rather than 'Minimum Height'. Never even thought of altering this one to get done what I wanted to do.

Have now increased this to 1.5mm and all is good in my garden!
[image]

Thanks to all, willow52

_Originally posted by @happycncer in https://github.com/martin2250/OpenCNCPilot/issues/28#issuecomment-719539900_

deHarro, 2020-10-31:
Hi Martin!

On changing texts...
How's about implementing a translation option?
I'm sure here are some guys that will help in translating to their native language.

And I myself may perhaps understand the one or ather topic better afterwards (when OCP speaks german to me)... ;-)

Harald

deHarro, 2020-11-01:
Hi Martin,

time would be no problem for me, but I tried to implement a second COM interface into your code some time ago and failed piteously.
So I have to admit, I'm not speaking C#, and I feel too old to try to get acquainted with it :)

Harald

martin2250, 2021-02-28:
Hi all,

I added some better explanations in 2a23fb2dcf19f6da9bc781af9301655d5c57f9d9.

Cheers!

closed 2021-02-28 by martin2250