martin2250/OpenCNCPilot #148: "Max Depth" in probing settings has no effect

deHarro, 2020-11-06:
Hi Martin,

I cannot realize any effect when changing the "Max Depth" in probing settings.
I think this should stop the descending of the Z axis, when there is no touch event after descending for that distance.

Or did I get that wrong?
Thanks!

Harald

martin2250, 2020-12-11:
Hi Harald,

sorry for the late reply. You're right, "Max Depth" should limit the plunge distance relative to Z=0. I'll look into this the next time I have the project open.

Thanks,
Martin

deHarro, 2022-05-08:
Hi Martin,
is this topic still pending?
Harald

martin2250, 2022-05-09:
Hi Harald,

sorry, I completely forgot about this. Will look into it soon.

Martin

deHarro, 2022-05-09:
Hi Martin,

don't hurry, I didn't need this feature over the past years and so didn't run into it.
I only thought you may have forgotten to close the issue (after having it fixed)...

Harald

martin2250, 2022-05-15:
Hi Harald,

I just tested it live on my machine and it works exactly as intended, eg stopping at Z=-5 when no contact is made.

about joystick support: That would be nice, but I tried something similar with a handwheel (~2-3 years ago) and I couldn't get it to work well. Since I currently don't have access to my machine often, this isn't going to happen anytime soon I'm afraid.

Cheers!
Martin

deHarro, 2022-05-15:
Hi Martin,
thanks for testing!
Ok, I will test it too, and if I get the same result, I will close this issue, ok?

As for your handwheel...
Did you implement a routine to communicate to the handwheel via another serial port?
If so, is this additional routine still implemented by chance? :)

I would make efforts in that direction on my own, but I didn't find any way into your code, when I tried that years ago.
Having your approach on hands as starting point would be easier for me...

Harald

deHarro, 2022-05-15:
Hi Martin,

I just tested probing with OCP V1.5.12 and found, probing stops at the given depth, without contact.
So this issue is closed.

Harald

closed 2022-05-15 by deHarro

martin2250, 2022-05-16:
Hi Harald,

I think I stated a separate branch for that, but I can't find it here on GitHub. Maybe it's buried somewhere on an old hard drive.

Cheers!

deHarro, 2022-05-16:
Hi Martin,
no big thing. If you happen to stumble across it, think of me... 😉
Thanks!
Harald