martin2250/OpenCNCPilot #149: Use As a Post Processor Or Send Over WiFi

reynolds087, 2020-11-13:
I am using a different flavor of GRBL, (bdring/GRBL_ESP32) for the esp32 microcontroller. I currently control the CNC over WiFi using ESP3D and run my programs from SD card connected to the ESP32. Is there any way to use this program to post process my gcode files without having an active serial connection to the CNC? Ideally, I would want to do the probing over the WiFi connection and then use the software to post process the file, or even manually enter the grid points into an array and then post process offline, if there isn't a way to do it using the WiFi connection.

martin2250, 2020-12-11:
Hi,

sorry for the late reply. OpenCNCPilot needs an active serial connection for probing. You could, however, generate a height map, save it to disk, edit the points manually and apply that to a gcode file.

Martin

reynolds087, 2020-12-14:
Thanks, 
when I create a height map and save it to disk, it appears to generate an XML file with only this line inside:
`<heightmap MinX="0" MinY="0" MaxX="100" MaxY="50" SizeX="21" SizeY="11" />`

I can see the dimensions of the board and the number of points are defined in that first line, but I'm not sure how to create the elements for each individual point and their corresponding Z coordinates.

martin2250, 2020-12-16:
When creating a new height map, you can choose to generate points based on a mathematical formula (at the bottom of the dialog). Just enter "0" as the formula and all points will be present and set to zero.

gonzalocargut, 2021-01-13:
I am currently modifying this program to support a TCP connection, either wifi or lan, at the moment it works using ESP8266 as a converter (to which plane integrate buttons to do local jog), I must try it for a while to approve it and upload the version to github.

Is it something like what you needed? With this you will control the machine completely by ethernet.

reynolds087, 2021-01-13:
> I am currently modifying this program to support a TCP connection, either wifi or lan, for the moment it works using ESP8266 as a converter (to which I plan to integrate buttons to do local jog), I must try it for a while to consider it approved and upload the new version to github.
> 
> Is it something like what you needed? With this you will control the machine completely by ethernet.

Yea, exactly!! That would be great. Is there any documentation about using the esp8266 in the meantime?

gonzalocargut, 2021-01-13:
In ESP implement a simple bridge for now, I take what comes through the connection and send it through the serial port and vice versa, it is very simple, I still do not add the rest of the features that I want it to support, once I have it more or less advanced I will publish the source code in addition to the circuit to be assembled, I have to get some components from China.

gonzalocargut, 2021-01-13:
> Yea, exactly!! That would be great. Is there any documentation about using the esp8266 in the meantime?

In ESP implement a simple bridge for now, I take what comes through the connection and send it through the serial port and vice versa, it is very simple, I still do not add the rest of the features that I want it to support, once I have it more or less advanced I will publish the source code in addition to the circuit to be assembled, I have to get some components from China.

martin2250, 2021-02-28:
Hi all,

I merged Gonzalo's PR, so you should be able to use the ESP32 wirelessly now. Let me know if there are any issues.

Cheers
Martin

closed 2021-02-28 by martin2250

reynolds087, 2021-04-12:
I know this is pretty late, but just wanted to thank you guys for working on this. I finally got my CNC up and running after a rebuild, and now am starting to use this feature. I really appreciate the work you both put in. Thank you!!!