martin2250/OpenCNCPilot #150: Big amount of ram over 500.000 line

Elvis999IT, 2020-12-09:
Hi
I use big file, with 500.000 /700.000 line
After few hours of perfect work the machine begin to stop for few second every few second.
I noticed that the usage of ram of the pc increase from begin of the job until it reach the max amount after few hours and the start stop begin.
Could be a opencncpilot issue?
Thanks

martin2250, 2020-12-11:
Hi Elvis,

I thought I'd fixed all memory leaks... Can you provide the gcode file for reference so I can try to reproduce this? Also how much RAM are we talking about (1: after loading the file and 2: when the stuttering begins)?

Thanks,
Martin

Elvis999IT, 2021-01-09:
Hi
Sorry if i answer only now but i didn't hava chance make some test before
1- Opening OpenCNC - Ram 32 Mb
2- Load Project - Ram 1.3 Gb
3- During work ram usage keep encrease slowly
4- After 1 hour - Ram 1.7 Gb Keep increasing
5- After few Hours 2.9 Gb
6- Beging stuttering
 
My laptop:  
Win 7 Home Premium SP1 64 bit
Intel i5-2410M 2.30 GHz
Ram 4.0GB
[02 Sgross 3mm.zip](https://github.com/martin2250/OpenCNCPilot/files/5791115/02.Sgross.3mm.zip)

Thanks
Elvis

martin2250, 2021-01-31:
Hi Elvis,

I looked into this issue and I couldn't find any obvious errors. When running this program on an "accelerated" grbl (movements happen instantaneously), the memory usage increased only very slowly (at a similar speed to what you're describing). So the RAM isn't used up in proportion to the number of lines sent. This makes this problem fairly hard to track down.

This is an uncommon use for OpenCNCPilot, so I'd say for this purpose you'd be better off using a different gcode sender, that doesn't have such a complicated visualizer. On my fairly ok PC, the viewport was very laggy, so even if the RAM usage didn't keep increasing, I wouldn't recommend OCP for files as large as yours.

If you need OCP for probing, you can always export the gcode after applying the height map and use a different gcode sender.

Best,
Martin