martin2250/OpenCNCPilot #152: x y autolevel offset

roneylf, 2021-02-15:
there is no offset of x and y as in bCNC

martin2250, 2021-02-15:
please elaborate, what does this setting accomplish?

roneylf, 2021-02-15:
it would be nice if there was the option to offset the X and Y axes for the autolevel, in my CNC the auto level is not in the same place as the Drill, I speak of having something like the example in the image

[image]

martin2250, 2021-02-28:
Okay, I'll mark it as nice-to-have. When I find some free time, I'll implement it... or if someone wants to submit a PR - go ahead.

Until then, you can always move your coordinate system after probing, which shouldn't be too much of a hassle.

deHarro, 2023-10-07:
Sorry for reusing this issue, but it fits perfectly ;)

I read about the implementation of this offset feature, but at first was unable to find it in the gui.
Intuitively I looked for the offset controls in the settings-probing dialog, but in vain.

After looking up the code I found, the addition is in deed part of the code as stated on Github, so I again digged into the gui, and finally I found the controls in the "manual Probing" Tab.

As those values are unlikely to frequent changes, I would prefer to move the settings to the settings dialog, where I first searched for.

Just my 5 cent.

Harald

martin2250, 2023-10-07:
Whoops, I completely missed that @Paciente8159 didn't move the inputs to the settings window as requested and moved it there instead :/ I'll move it again...