martin2250/OpenCNCPilot #153: Добавление новых функций

max-radio, 2021-02-28:
Добрый день, планируется ли добавление позиционирования станка по usb камере? А также поворот и трансформация УП по реперным точкам, компенсация люфта по осям, выставление геометрии станка (если станок кривой)?
Давайте реализуем все возможности planet cnc, mach3, и т.д. Я могу выступать в качестве тестера.
Очень понравилась ваша программа, но нет тех функций которые я хотел бы видеть. Надеюсь проэкт ещё жив и будет развиваться.
Спасибо.

martin2250, 2021-02-28:
Good evening (here in DE),

most of these things are nice to have, but I don't think that they would add too much value without introducing a lot of complexity. There is an interesting fork with some of these features: https://github.com/bytemaster-0xff/OpenCNCPilot
For backlash compensation, I'd rather look into some of the grbl forks for STM32.

In general, OpenCNCPilot is at a state where it is rather tedious to add large new features. I'd like to rewrite it completely someday, but I don't have that kind of time at the moment.

Cheers,
Martin

max-radio, 2021-02-28:
Я так не думаю. Эти функции очень даже полезны, и только из-за них контроллер planet cnc стоит 300евро.

max-radio, 2021-02-28:
Тогда вопросик, а в какой программе или на каком языке писали саму программу? Если не хотите сами, попробую я.

martin2250, 2021-03-01:
You may be right about these features being useful, but I don't want to invest that much time into features I don't need. If you want to implement these features, I'd be happy to accept the pull request.

closed 2021-03-01 by martin2250