martin2250/OpenCNCPilot #157: Can I add support for grbl-Mega-5X?

viewsat, 2021-04-21:
Hello there:
I am using grbl-Mega-5X( https://github.com/fra589/grbl-Mega-5X/issues/192 ), but it is not compatible with OpenCNCPilot. I hope that OpenCNCPilot can increase the support for grbl-Mega-5X in XYZ three-axis
thank you!

martin2250, 2021-04-21:
Hi,

do you know the differences between this and regular grbl that make it incompatible? I don't have enough time to do this myself currently, but if you  want to tackle it yourself, I'm looking forward to the pull request :wink:

Cheers!

fra589, 2021-04-23:
Hi,  
grbl-Mega-5X is fully Gcode input compatible with the original grbl and have similar output when using it with 3 axis XYZ.  
The main differences are that grbl-Mega-5X can drive up to 6 axis, the axis can be renamed in the list : X, Y, Z, A, B, C, U, V and W. So, the interface program must be able to use these letters as axis names.
Axis can also be cloned, using one axis name for more than one axis, for example you can have an XYZ system with 4 axis used : XYZX. So the default output of grbl-Mega-5X will have more than 3 axis coordinates output, for the example:
```
?
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0|WCO:-100.000,0.000,0.000,0.000,0.000>
```
Unless you use the ```#define REPORT_VALUE_FOR_AXIS_NAME_ONCE``` sentence in the config.h grbl-Mega-5X file.
  
@++;  
Gauthier - grbl-Mega-5X maintener

martin2250, 2021-04-23:
Hi Gauthier,

in that case it should already be possible to use it. In the settings window there is an option called "Ignore Additional Axes in Status Messages". That should make OCP work with 4+ axis machines.

Best, Martin

viewsat, 2021-04-24:
> Hi Gauthier,
> 
> in that case it should already be possible to use it. In the settings window there is an option called "Ignore Additional Axes in Status Messages". That should make OCP work with 4+ axis machines.
> 
> Best, Martin

Thank you very much!
I have used grbl-Mega-5X according to your method today and the result is ok.
I will actually carve the pcb to see the result

closed 2021-04-30 by martin2250