martin2250/OpenCNCPilot #158: Height map going too deep or too shallow

liquibyte, 2021-07-16:
I've been trying to get a pcb cut but can't seem to figure out how to keep the bit from going either too deep or too shallow.  I want to make a cut 0.005mm deep (a 2oz boards copper is 1.4mil or 0.03556mm thick) but when I apply a height map the z axist gets set to 0.3+ mm.  I don't understand what I'm doing wrong here.  I'd also like to add that there are a couple of features in bCNC that would be nice to have.  The scan feature is nice to make sure your bit will stay within the bounds of the workpiece.  The feature of the bit going to 0,0 at xy after probing seems like it would be nice too.  I may be missing this but a single probe to determine z axis 0 at xy 0,0 seems like a must have feature.  Alas, I can't seem to figure out how bCNC even works, it looks like it was programmed by someone from the early 90's.

martin2250, 2021-07-19:
Hi Michael,

OpenCNCPilot intentionally doesn't probe Z=0 at the start of a run so you can set the zero position at an arbitrary point. (XY zero might get milled away in the middle of the process).

Your issue sounds a bit like you probe the board and then afterwards move your coordinate system. Did you set Z=0 after probing?

Cheers, Martin

silicon-systems, 2021-08-10:
> Hi Michael,
>
> OpenCNCPilot intentionally doesn't probe Z=0 at the start of a run so you can set the zero position at an arbitrary point. (XY zero might get milled away in the middle of the process). [...]

Hi All,

Can we be at least kind enough to leave a reply if an issue was resolved or not.

Hanspeter.