martin2250/OpenCNCPilot #159: Z-Axis driven to top limit when Heightmap run

charlt0n, 2021-09-17:
Using GRBL file from KiCad into FlatCam to produce the .nc file for a PCB design.  Trying to use the probing function in OpenCNCPilot but finding that running the Height Map function immediately takes Z axis to its top limit even after successfully establishing the correct initial probe contact height.  The CNC machine is a Sainsmart 3018 ProVer.  Would appreciate help in determining what to check for misconfiguration.  The HOME and set probe Zero functions work correctly.

charlt0n, 2021-09-17:
Found the problem after finding additional online docs.  Great application & thanks..

martin2250, 2021-09-21:
Hi @charlt0n 

glad you figured it out! What was the cause of the issue? I'm sure someone else will also find this useful in the future :)

Cheers, Martin

charlt0n, 2021-10-02:
Martin, it was my own fault - not manually sending the coordinates after I did the probe detection of the board height baseline - in other words, not following he directions!