martin2250/OpenCNCPilot #160: when click "Probe and set zero" Z-axis motor vibrates and does not move

rbb-soft, 2022-04-12:
Hi there and thanks for using OpenCNCPilot,

before you submit this issue please make sure to check the following things:
- you're using the latest version of grbl (1.1f)
- you have the most recent version of the [.NET Framework](https://www.microsoft.com/net/download/dotnet-framework-runtime) (minimum 4.7)

If you still need help, delete this text and describe your problem and I or another member of our small community will help you.

rbb-soft, 2022-04-12:
hi im using 1.1h version

martin2250, 2022-04-13:
Hi,

I didn't see any relevant changes in the changelog, so it probably isn't the new version of grbl. Can you try what happens when you send the following command:
```
G38.2X10F100
```
This shoud just move the x-axis by 10mm, unless interrupted by the probe pin. It that works, try the same thing with Z.

rbb-soft, 2022-04-13:
Fixed by installing version 1.1f.

Thank you very much for your time and for such a fabulous application

martin2250, 2022-04-14:
Ok, that's unexpected but not unwelcome. I don't think you're missing out on much by using 1.1f instead of h.

You're welcome, have fun using OCP :)

closed 2022-04-14 by martin2250