martin2250/OpenCNCPilot #166: Fusion "no motion mode active" G28 G91 Z0

Levi--G, 2022-05-15:
Hi, :)
First: Wonderful project! :D

Im trying to send a fusion360 generated file to openCNCPilot, im getting "no motion mode active" in the parser on the line G28 G91 Z0.
I think it is because it doesn't support the Z-only homing yet with G28? is it possible to add that?

Thanks! <3

martin2250, 2022-05-16:
Hi Levi,

thanks! :)

Yeah, there is no direct support for G28 (also G53, or any other coordinate systems). I also like to have something like "G53 G0 Z-1" at the end of my Fusion G-Code, which gives me the same warning.

Supporting different coordinate systems is not something I plan on adding, since it doesn't make sense to change coordinate systems during PCB milling and it would be a lot of work.

That said, this warning doesn't stop you from running the g-code as it is, you just can't use the editing functions and height maps. If you want to use height maps, you'll have to change the line to not use G28.

Cheers!
Martin

Levi--G, 2022-05-16:
Hi Martin,

Ofcourse i get that :) the main issue for me is the previewing also doesn't work with the error, which is kindof a bummer since i really like that about OpenCNCPilot. :D
For now i compiled my own version with this in the parser:
´´´
            if (line == "G28 G91 Z0")
            {
                var EndPos2 = new Vector3(State.Position.X, State.Position.Y, 0);
                State.PositionValid[2] = true;
                Line motion = new Line();
                motion.Start = State.Position;
                motion.End = EndPos2;
                motion.Feed = State.Feed;
                motion.Rapid = true;
                motion.LineNumber = lineNumber;
                motion.StartValid = State.PositionValid.All(isValid => isValid);
                State.PositionValid.CopyTo(motion.PositionValid, 0);
                Commands.Add(motion);
                State.Position = EndPos2;
                return;
            }
´´´
But ofcourse that is very hacky... Would it instead be a possibility to add a series of lines to ignore in the parser as a part of the program settings? so i can use the normal version? :) or even exchange them just for the parsing, but send the original to the cnc? Just some ideas...

Friendly regards,
Levi

martin2250, 2022-05-18:
I think I did something similar in the https://github.com/martin2250/OpenCNCPilot/compare/passthrough branch. Not sure why it is abandoned... I can look into this some time