martin2250/OpenCNCPilot #171: start Probe error

Plovmann, 2023-02-22:
Hello, Martin.

I'm fairly new to your tool, and wow, its amazing.
However, I have a problem. I want to engrave something in aluminum.
 Right from the start, load my file, and puts the probe clamps on the board - bit, presses create new and presses run, then bit doesn't go down and hits.. it only goes down a bit and then it starts with doesn't hit the board and doesn't write PROBE with red.

Then I have to adjust the bit ALL the way down to the plate before it will run correctly. Why ?

If I run the same file, with the same start, several times I get different depths, why?

Do you have to drive down the bit from the start and adjust your Z height? just mean that's what it should go down and do from the 1st probe, right?

I hope you understand what I mean

Regards Lars

martin2250, 2023-02-22:
Hi Lars,

I'm not sure if I understood everything, but it sounds to me like you start the probe cycle too high up in the air.

Before pressing "Run", you should set up your work coordinate system so that Z=0 is very close to the surface. There is a limit to how deep OpenCNCPilot will go at most, which is 5mm down relative to the origin (Z0).

You can either set the origin by eye, maybe 1-2mm above the surface, but then probing will take longer and changing tools is harder. I recommend probing the first point manually (using G38.2) and setting the origin exactly on the surface, that way you can also use this position to touch off other tools later.

Let me know how that goes, Cheers!
Martin

Plovmann, 2023-02-22:
Hi Martin, and thanks for the quick reply.

Ok, Is there a reason why you couldn't set it to 10 mm and it still runs down to the plate?

Does that mean, when I now drive it down to 1-2 mm. above the surface, then I have to reset Z by running G10 ?

I haven't done that, could that be why I get different depths different times (with the same file)

Will you explain to me how I insert (and where) G38.2?

Is it than possibly after engrave I can move up Z, change tools and move down to Z0 again and cut out the plate?

Is there a function where it automatically drives Y,X,Z down to 0 again?

Last wonder :)
how can it be that start has to be pressed twice for it to start?

 But a great program, and I simply had no idea you could do it that way. Stumbled across it on google :)

Have considered making a probe with some form of microswits that can be mounted on a spindle, in order to be able to use it on wood :)

Mit freundlichen Grüßen
Lars Plovmann

martin2250, 2023-02-23:
Hi Lars,

> Ok, Is there a reason why you couldn't set it to 10 mm and it still runs down to the plate?

I limited the maximum depth to prevent crashing the machine in case anything goes wrong

> Does that mean, when I now drive it down to 1-2 mm. above the surface, then I have to reset Z by running G10 ?

yup, either that or G92

> Will you explain to me how I insert (and where) G38.2?

move the machine 1-2 mm above the surface, set the origin, then use "G38.2 F20 Z-5" to probe the surface at that point. You can skip setting the origin if you adjust the "-5" accordingly.
When the probing cycle has finished, set your origin with G10 or G92.

> Is it than possibly after engrave I can move up Z, change tools and move down to Z0 again and cut out the plate?

yes, just use G38 and G92 again to adjust for the change in tool length.

> Is there a function where it automatically drives Y,X,Z down to 0 again?

no, that would be dangerous because it could crash the machine

> Last wonder :)
> how can it be that start has to be pressed twice for it to start?

umm, that's new... is the window not in focus the first time?

> But a great program, and I simply had no idea you could do it that way. Stumbled across it on google :)

Thanks :)

> Have considered making a probe with some form of microswits that can be mounted on a spindle, in order to be able to > use it on wood :)

People have done that, you can find something in the Github issues.

Grüße
Martin

Plovmann, 2023-02-23:
Thanks alot Martin...  I will try and see :)  I promice to went back and tell about it :) 

// Lars

Plovmann, 2023-03-02:
Hey again.  
I get it all working.. Im a happy man.  Thanks alot for your support.

Mit freundlichen Grüßen
Lars Plovmann
Wir haben Deutsch in der Schule hier in Dänemark ;)

Plovmann, 2023-05-17:
Hallo Again Martin. 
Hmm, I still have a issue - problem about the program, and made a little video to explaine whats my problem is.. Is there a place where I can send it to you ?  its pretty big ( 150 mb ) so perhaps weetransfer or ??

Regards
Lars Plovmann

martin2250, 2023-05-18:
Hi Lars,

I'm fine with any platform as long as I don't have to sign up to view the video, vimeo/youtube/Gdrive/dropbox/weetransfeer is fine.

Cheers

Wiki-the-Tinkerer, 2023-08-23:
Hi, Plovmann,
the "how can it be that start has to be pressed twice for it to start?"
may very well be the same thing i wondered about. 
The "send-start" sends the code, AND starts moving the machine. 

Then, if there is a "pause" in the code  (e.g. waiting for a tool change, or similar), it stops cutting, but does not say "pause" on the "send the code" side. Then you have to press "Start -cut" on the center menu to continue the cut (even if the other Start-send already started cutting). 
Can be difficult to know whats going on, especially if there are many pauses. 

If this is the case, look for M0 or M1 on the code - it highligths it the "File" menu (but often not at a visible point, so scroll). And press the other start.